qmk/qmk_firmware #2918: Change hamburger menu placement

skullydazed, 2018-05-08:
As pointed out on reddit we need to change how the hamburger menu works: https://www.reddit.com/r/olkb/comments/8huf00/qmk_proposed_replacement_for_gitbook_feedback/dyn8qoi/

We're looking for a frontend developer who can either work some CSS magic or write a docsify plugin to make that menu behave more sanely.

aergonaut, 2018-05-10:
Looking at the CSS in `qmk.css`, the hamburger has `position: fixed` and `bottom: 0`, which is what sticks it to bottom. Changing to `top: 0` instead would stick it to the top, but that leads to the hamburger overlapping the search field:

[image]

You could shift the search bar down, but that ends up leaving about 40px (I eyeballed) of negative space:

[image]

I thought the negative space would look stranger, but after trying it out, it doesn't really look that weird. So that could be an option.

The CSS for when the sidebar is collapsed seems to just leave a gutter where the hamburger sits, so moving it up to the top doesn't seem to affect the usability of the site in that state.

aergonaut, 2018-05-11:
@skullydazed yes, I'd be happy to make a PR. I'll need to read a little about how docsify works, since I've never used it before, but I should be able to get a PR up for you in the next couple of days.

PR #2956 (2018-05-14): Override hamburger placement
Overrides hamburger button and search bar placement in the sidebar. Fixes #2918.

[image]
closed 2018-05-14 by skullydazed