qmk/qmk_firmware #2932: Change the value of action_get_macro while holding

baronTommy, 2018-05-10:
Hold `M(0)`
Expected results
`through` `A`, `A`, `A`, `A` ...

```c
static bool through_flg = true;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {
  if (record->event.pressed) {
    switch(id) {
      case 0:
        if (through_flg) {
         //  !! Change the value of action_get_macro while holding
          through_flg = false;
          return;
        }
        SEND_STRING("A");
        break;
    }
  }
  return MACRO_NONE;
};
```

drashna, 2018-05-10:
What are you trying to do specifically?

baronTommy, 2018-05-11:
@drashna 
Hello 

karabiner __BlockUntilKeyUp__  like
https://github.com/tekezo/Karabiner/blob/9950919eb7fc6fe631b61cf8acdc409aa93ed6f4/src/core/server/Resources/include/checkbox/vi_mode.xml#L48

```
M(0) Tap
    through (key down)
    A (key up)
```

```
M(0) Hold
    through (key down)
    A .... (repeat)

Behavior similar to X LT(layer, kc)
The difference is that kc repeat
```

```
M(0) down (through)
↓
K down (5 send) ten key like
K up
↓
J down (4 send) ten key like
J up
↓
M(0) UP (through)
```

```
M(0) down (through)
↓
M(0) UP (A send)
```

drashna, 2018-05-11:
I'm still not sure what you want to do....

but I think you want `register_code`, and `unregister_code` here, rather than `send_string`.  And you'd want to use a timer, which is how LT works.

baronTommy, 2018-05-11:
@drashna 

Is it feasible?

```
LT(layer, kc)

Tap
    send kc

short hold
    layer switching

long hold
    send kc
    send kc
    send kc
    send kc
    ...
```

drashna, 2018-05-11:
Well, it's a bug right now, but with the LT code, if you tap, release and then hold it (all within the "tapping term"), it will repeatedly send the keystroke (well, no, it won't, but it will register as down, and the OS will handle it... which will repeat it).

closed 2018-05-14 by baronTommy