qmk/qmk_firmware #2948: Help with hex, msys64, make command not found

neanic, 2018-05-12:


drashna, 2018-05-12:
make sure that you are actually in the qmk folder.

You may need to run `cd qmk` or `cd qmk_firmware` before running the make command.

neanic, 2018-05-14:
I've tried a few different folders with the same result. I'm pretty sure I'm in the right one though. 
I had a similar message earlier in the install when I was trying to run a git command. I ended up having to run this command:  pacman -S git﻿ . I was wondering if there was something like that that I'm missing. It took me a while to find that and I haven't been able to find anything, except what you recommended, about it.

neanic, 2018-05-14:
Thanks for trying guys. I think I'm just too dense.  I followed everything, to the best of my ability, with the same result. 
./util/qmk_install.sh didn't seem to do anything...
Anyone want to buy a soldered iris with blue leds?

drashna, 2018-05-14:
If you are fine with using the default layout: http://qmk.fm/compiled/iris_rev2_default.hex
https://github.com/qmk/qmk_firmware/tree/master/keyboards/iris/keymaps/default/keymap.c

Otherwise, if you have a customized keymap that you want to use, you can link it here, and and I can see about helping. 

And I do have some remote support tools, so I could possibly help with gitting it installed.

neanic, 2018-05-14:
I deleted qmk and reinstalled it.

[image]

jbtompkins, 2018-05-15:
Have you installed gcc on msys2? I think the command is

`pacman -S mingw-w64-x86_64-toolchain`

neanic, 2018-05-15:
Installed it. 
Same result.

drashna, 2018-05-15:
try `util/msys2_install.sh`instead

neanic, 2018-05-16:
It works!
Thanks for all the help. I'll try to flash it tomorrow.

neanic, 2018-05-25:
Having issues flashing my pro micros. Should I start a new issue?

neanic, 2018-05-31:
[image]

I'm sure I'm doing something wrong. I also tried to do it in msys, but couldn't connect either.

drashna, 2018-06-01:
AVRDUDESS needs to use the "avr109" programmer, and use the PORT in question. 

[image]

neanic, 2018-06-02:
[image]

neanic, 2018-06-04:
Do I hold the button when turning it on or  push it after? Should I keep it pressed while flashing? Do I have to change the port manually somehow?

neanic, 2018-06-15:
It changes from com4 to com5 but still doesn't work.

neanic, 2018-06-19:
I think it may be a driver problem, but I'm not sure how to fix it.

neanic, 2018-06-23:
Thanks for the help. Sorry for being so dense.

closed 2018-06-23 by neanic