qmk/qmk_firmware #2960: How do I change the serial PIN for let's split

fi0, 2018-05-14:
Let's split uses PD0 for linking up left hand and right hand. How do I change it to PD3?

drashna, 2018-05-14:
it's defined in `serial.h`, IIRC

fi0, 2018-05-14:
I tried SERIAL_PIN_MASK _BV(PD3)
Either side of the keyboard works but the linkage doesnt. Do I need to change other variables too?

drashna, 2018-05-14:
the INT0 may need to be changed, too, but I'm not sure.

drashna, 2018-05-14:
https://camo.githubusercontent.com/f39d220b2fcd3eb1b5cc6721a6a275a31c1eedab/687474703a2f2f7777772e70696768697878782e636f6d2f746573742f77702d636f6e74656e742f75706c6f6164732f323031362f30372f70726f5f6d6963726f5f70696e6f75745f76315f305f7265642e706e67

Based on that image, yes, you'd need to use this: 
```c
#define SERIAL_PIN_INTERRUPT INT3_vect
```

I think....

fi0, 2018-05-14:
I found out i need to modify serial.c as well. Thanks for the tips @drashna

closed 2018-05-14 by fi0