qmk/qmk_firmware #2988: I'm still a bit confused with the make command. I want to compile a hex for my satan gh60 in the poker layout and when I do this  "satan:poker" it says make: *** No rule to make target 'satan:all'.

anorraq, 2018-05-17:


drashna, 2018-05-17:
What folder are you in?

anorraq, 2018-05-17:
I'm a bit confused with your question (first time really doing this stuff). If you mean which folder was the command window in, I hadn't entered any folder so I was entering the code. I did try to use "/c/qmk_firmware/keyboards/satan:poker" and still got the same result (if that was what you were wondering).

drashna, 2018-05-17:
For the linux shell, the folder matters.  You should be in the qmk_firmware folder.  And from there, you want to run `make satan:poker`. 

so ... for here, run `cd /c/qmk_firmware`, and then run `make satan:poker`

anorraq, 2018-05-18:
awesome it works now! Thanks for bearing with me.

drashna, 2018-05-18:
awesome, glad to hear it!

drashna, 2018-05-25:
Since this is solved, closing.

closed 2018-05-25 by drashna

speakeasymy, 2020-11-26:
this has helped me too. thank you!