qmk/qmk_firmware #3001: Serial hangs on Teensy LC with ChibiOS

micolous, 2018-05-19:
Whenever serial communication on QMK is used on Teensy LC with ChibiOS, the device appears to hang, regardless of whether there is something on the PC side listening to the serial port.  This impacts use in Steno mode.

I saw that this functionality was recently added in #2362 and #2750.

Steps to reproduce:

1. Adapt `chibios_test/teensy_lc_onekey` to emit some Steno keys ([my modifications found here](https://github.com/micolous/qmk_firmware/tree/virtser-hang/keyboards/chibios_test))
   * `MOUSEKEY_ENABLE`, `EXTRAKEY_ENABLE`, `CONSOLE_ENABLE`, `COMMAND_ENABLE`, `SLEEP_LED_ENABLE` = no
   * `STENO_ENABLE`, `VIRTSER_ENABLE` = yes
2. Build and flash the firmware on Teensy LC (using `teensy_loader_cli`)
3. Open a serial terminal emulator such that it will be reading from the Teensy's serial port.

I added some extra debug logic, so that the LED on the Teensy would flash whenever it detected a keypress.

* When in steno mode, the LED flashes for the first keypress, but not subsequent keypresses.

* When in non-steno mode, the LED flashes for any keypress.

* When commenting out the call to `chnWrite` in `virtser_send` and in steno mode, the keyboard LED flashes for any keypress.

* However, with the previous change (or without it), as soon as any data is sent by the PC to the Teensy over the serial link, the Teensy will lock up (and LED will not flash)

Host system was running Ubuntu 17.10 (Linux 4.13 kernel).

micolous, 2018-05-19:
Looking at a USB packet capture with `usbmon` in Wireshark, I see, when starting serial terminal or Plover:

- `SET CONTROL LINE STATE` is set to `0x02` (DTE, activate carrier control)

- Teensy responds with `URB_CONTROL` success

- `SET LINE CODING` is sent to the device with correct parameters (115200 8N1)

- Teensy responds with `URB_CONTROL` success.

I can't see any sign of messages when keys are pressed or when serial data is sent to the device.

I was able to get a `Keyboard start.` from the console intermittently using `hid_listen`, but I never got any debugging messages (when debugging was turned on).

micolous, 2018-05-19:
I've tried a couple of other things.

Using `chnWriteTimeout` doesn't help -- the device still locks up.

I managed to get a USB link level trace.  For this, I disabled the keyboard endpoint (it's very noisy), so the interface descriptor looked like this:

```
$ lsusb -d feed:6464 -v

Bus 003 Device 090: ID feed:6464  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0xfeed 
  idProduct          0x6464 
  bcdDevice            0.01
  iManufacturer           1 QMK
  iProduct                2 ChibiOS QMK test
  iSerial                 3 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           70
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         1
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        1
        bSlaveInterface         2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               5
Device Status:     0x0000
  (Bus Powered)
```

As soon as I connected to the serial port:

```
[        ]   2.724894 d=  2.602671 [ 82.0 +531.250] [  3] IN   : 10.3 
[        ]   2.724897 d=  0.000003 [ 82.0 +534.583] [  1] NAK 
[        ]   2.724905 d=  0.000008 [ 82.0 +542.333] [  3] SETUP: 10.0 
[        ]   2.724908 d=  0.000003 [ 82.0 +545.500] [ 11] DATA0: 21 22 03 00 01 00 00 00 7f ed 
[        ]   2.724917 d=  0.000009 [ 82.0 +554.167] [  1] ACK 
[        ]   2.724931 d=  0.000014 [ 82.0 +567.917] [  3] IN   : 10.0 
[        ]   2.724934 d=  0.000003 [ 82.0 +571.250] [  1] NAK 
[        ]   2.724951 d=  0.000017 [ 82.0 +587.917] [  3] IN   : 10.3 
[        ]   2.724954 d=  0.000003 [ 82.0 +591.250] [  1] NAK 
[        ]   2.724976 d=  0.000022 [ 82.0 +613.000] [  3] IN   : 10.0 
[        ]   2.724979 d=  0.000003 [ 82.0 +616.333] [  3] DATA1: 00 00 
[        ]   2.724982 d=  0.000003 [ 82.0 +619.833] [  1] ACK 
[        ]   2.725011 d=  0.000029 [ 82.0 +648.500] [  3] IN   : 10.3 
[        ]   2.725014 d=  0.000003 [ 82.0 +651.750] [  1] NAK 
[        ]   2.725056 d=  0.000042 [ 82.0 +693.500] [  3] IN   : 10.3 
[        ]   2.725059 d=  0.000003 [ 82.0 +696.750] [  1] NAK 
[        ]   2.725101 d=  0.000042 [ 82.0 +738.500] [  3] IN   : 10.3 
[        ]   2.725104 d=  0.000003 [ 82.0 +741.750] [  1] NAK 
[        ]   2.725145 d=  0.000041 [ 82.0 +782.500] [  3] IN   : 10.3 
[        ]   2.725148 d=  0.000003 [ 82.0 +785.750] [  1] NAK 
[        ]   2.725190 d=  0.000042 [ 82.0 +827.500] [  3] IN   : 10.3 
[        ]   2.725193 d=  0.000003 [ 82.0 +830.750] [  1] NAK 
```

Looks like the CDC ACM handshake is done, then the host never gets any data out of the Teensy.  Even when the device should be sending data, there's no sign of it in the trace.

micolous, 2018-05-26:
As an experiment, I tried reverting e9d32b6 (PR #2750), and the device stops showing up at all on USB if `STENO_ENABLE = yes`. Pressing the program button on the Teensy LC still works to pull it into programming mode.

I've also loaded other programs using Teensyduino -- they're fine, and able to use the USB serial device.

My current suspicion is that there's an issue with the interrupt handler in QMK's USB driver, and whenever something needs to touch those endpoints, it lands in a deadlock.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc