qmk/qmk_firmware #3005: MT() multiple modifiers doesn't work

darutse, 2018-05-19:
#define T1 MT(MOD_RALT, KC_ENTER) - this one works it activates RALT if held and sends Enter if tapped
#define T2 MT(MOD_RALT | MOD_LSFT, KC_SPACE) - this one doesn't work I just need to add Shift to Alt when holding the key.
If I hold T1 and manually add shift it works. What am I doing wrong here?

drashna, 2018-05-19:
Works for me.   But it uses right alt and shift. 

`MT(MOD_RALT|MOD_LSFT,KC_SPC)`
Also, make sure your repo is up to date.

darutse, 2018-05-20:
OK I finally figured this out. First, a small intro on how and why I use these. I use two thumb buttons (split spacebar) one key is for num layer or space if tapped, and second button for punctuation layer or enter if tapped.
But I could not just use default keycodes to make these layers, because I use different language layouts and each layout is linked to QWERTY in the OS (I use linux btw) So if I remap my QWERTY layout (in Xorg) it affects each other layout key positions. So it becomes sort of entangled mess (plus the fact that firmware only sends keycodes, not keys themselves, which greatly limits your factual multi language layout mappings)
I decided to use level3 and level4 OS layers to map these layers, it gave me more control through Xorg layout configuration file. For that I've used lvl3 and lvl4 shortcuts, in linux that's ralt and shift + ralt. Now the problem was my X configuration for keyboard (I use arch and openbox without DE so my keyboard config lays here `/etc/X11/xorg.conf.d/60-keyboard-layouts.conf`)
I've used `lv3:ralt_switch_multikey` as an option for X and it was wrong, you need just `lv3:ralt_switch` without multikey. Plus in X layout config file you should not use `include "level3(ralt_switch)"` at the end of your mappings.
That's all, it would make sense for those who do things similarly to me, bottom line is - not a bug, OS configuration issue.

closed 2018-05-20 by darutse