qmk/qmk_firmware #3021: Serial polling delay on Orthodox

amellors, 2018-05-22:
There appears to be a polling delay of the slave have on Orthodox (and maybe other Let's Split) based boards when building with Linux or OSX tool chain.

I have built and programed my Orthodox board with the default layout using .hex files from both an OSX based build and a windows (MSYS) based build.  There is no noticeable lag when using the build from Windows, how ever when I use the build from my OSX based machine the polling frequency of the slave board seems to be significantly longer.  So much so that if I tap on a key fast enough on the slave side I will get no input on the master. 

I will post the two different .hex files, if it matters, sometime.

drashna, 2018-05-22:
Seeing as most of the split boards share this code, it is likely an issue hta is affecting them, as well.

colinta, 2018-08-14:
@amellors just a thought, but could the versions of avr-gcc be different?

colinta, 2018-08-14:
I'm having this exact same issue (on a BFO-9000) but I don't have windows handy to verify that part.

amellors, 2018-08-14:
I'm not sure, I haven't built the firmware in a while.  But if both are using the managed toolchains shouldn't it be the same?

drashna, 2018-10-22:
@amellors I have a PR that updates the avr-gcc version on MSYS2.  It may be worth checking out. 
https://github.com/qmk/qmk_firmware/pull/4106

drashna, 2019-03-29:
mtei found an issue with serial communication based on the AVR GCC version. That may be the issue you're seeing actually.

Also, I have a PR to convert the Orthodox to the Split Keyboard code, which would address this issue.

colinta, 2019-03-29:
I think you’re right. Back when I was working on this I managed to get someone to compile the firmware with another gcc version (and on windows) and that resolved it I think - or maybe it helped but wasn’t entirely fixed? I ended up creating an entirely new firmware (github.com/colinta/Mechy).

drashna, 2019-04-05:
Well, that's one way to fix it.

That said, I have a PR to merge the Orthodox into the Split common code. That includes a number of improvements that mtei has added for serial.  So it may be worth testing that out too

98WuG, 2020-03-14:
@amellors @drashna Did this end up being resolved? I've been running my Orthodox on I2C for the past 2 years because of this and forgot to check up on the status of this issue.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc