qmk/qmk_firmware #3027: Unsure how to use Pro Micro Driver

HappyTreesDev, 2018-05-23:
Hello, I am trying to set up my keyboard using a pro micro. I can't seem to get the pins named correctly. I can't really find a good mapping of the arduino names to the atmega names and I don't understand how to use the driver.

drashna, 2018-05-23:
Ask and you shall recieve: https://camo.githubusercontent.com/f39d220b2fcd3eb1b5cc6721a6a275a31c1eedab/687474703a2f2f7777772e70696768697878782e636f6d2f746573742f77702d636f6e74656e742f75706c6f6164732f323031362f30372f70726f5f6d6963726f5f70696e6f75745f76315f305f7265642e706e67

HappyTreesDev, 2018-05-23:
@drashna I found a picture like this, but I am still unsure which names I should use? Do I need to do anything else other than using the names given here once I figure out which of these I would use?

drashna, 2018-05-23:
That really depends on the context/usage. 

But for the most part, the "Port" pins (tan colored), but drop the "P" in the name.

So for example the "TX0" pin (as labeled on the controller) would be "D3", since you drop the "P" from "PD3".

HappyTreesDev, 2018-05-23:
Hmm, I seem to have some other issue then. I did do that as I expected, but 00 on my chip seems to be mapping to 72.

Any idea what would cause that?

HappyTreesDev, 2018-05-23:
Actually, I think it has to do with how my keymap file is made. I am following a different example to make it now. Thanks for pointing me in the right direction and keeping me from going down the wrong rabbit hole.

closed 2018-05-23 by HappyTreesDev

drashna, 2018-05-23:
Well, if you're trying to build the board, it may help to use KLE and kbfirmware to build it initially.