qmk/qmk_firmware #3033: Linking fails when using gdisp image

pointlessone, 2018-05-23:
Hi,

Disclaimer: I'm an amateur C programmer.

I'm developing a custom keymap for ErgoDox Infinity. I got it working and decided to play with visualizer.

Basically, instead of text (as in devisualizer) I want to draw a few images. I think I've got my gfxconfig right. It compiles but the linking fails.

Here's my frame function:

```C
bool po_lcd_keyframe_display_layer_bitmap(keyframe_animation_t* animation, visualizer_state_t* state) {
    gdispClear(White);

    gdispImageOpenMemory(&img, &pixels);
    gdispImageDraw(&img, 10, 10, 8, 8, 0, 0);
    return false;
}
```

And here's the error:

```
Linking: .build/ergodox_infinity_pointlessone.elf                                                   [ERRORS]
 |
 | .build/obj_ergodox_infinity_pointlessone/keyboards/ergodox_infinity/keymaps/pointlessone/visualizer.o: In function `po_lcd_keyframe_display_layer_bitmap':
 | qmk_firmware/keyboards/ergodox_infinity/keymaps/pointlessone/visualizer.c:125: undefined reference to `gfileOpenMemory'
 | qmk_firmware/keyboards/ergodox_infinity/keymaps/pointlessone/visualizer.c:125: undefined reference to `gdispImageOpenGFile'
 | qmk_firmware/keyboards/ergodox_infinity/keymaps/pointlessone/visualizer.c:126: undefined reference to `gdispGImageDraw'
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/ergodox_infinity_pointlessone.elf] Error 1
make: *** [ergodox_infinity:pointlessone] Error 1
```

Can someone please help me fix this?

<details>
<summary>gfxconfig.h</summary>

```C
/**
 * This file has a different license to the rest of the uGFX system.
 * You can copy, modify and distribute this file as you see fit.
 * You do not need to publish your source modifications to this file.
 * The only thing you are not permitted to do is to relicense it
 * under a different license.
 */

/**
 * Copy this file into your project directory and rename it as gfxconf.h
 * Edit your copy to turn on the uGFX features you want to use.
 * The values below are the defaults.
 *
 * Only remove the comments from lines where you want to change the
 * default value. This allows definitions to be included from
 * driver makefiles when required and provides the best future
 * compatibility for your project.
 *
 * Please use spaces instead of tabs in this file.
 */

#ifndef GFXCONF_H
#define GFXCONF_H

///////////////////////////////////////////////////////////////////////////
// GOS - One of these must be defined, preferably in your Makefile       //
///////////////////////////////////////////////////////////////////////////
//#define GFX_USE_OS_CHIBIOS                           TRUE
//#define GFX_USE_OS_FREERTOS                          FALSE
[... 265 lines cut ...]
#define GFX_USE_GAUDIO                               FALSE
// There seems to be a bug in the ugfx code, the wrong define is used
// So define it in order to avoid warnings
#define GFX_USE_GAUDIN                               GFX_USE_GAUDIO
//    #define GAUDIO_NEED_PLAY                         FALSE
//    #define GAUDIO_NEED_RECORD                       FALSE

///////////////////////////////////////////////////////////////////////////
// GMISC                                                                 //
///////////////////////////////////////////////////////////////////////////
#define GFX_USE_GMISC                                TRUE

//#define GMISC_NEED_ARRAYOPS                          FALSE
//#define GMISC_NEED_FASTTRIG                          FALSE
//#define GMISC_NEED_FIXEDTRIG                         FALSE
//#define GMISC_NEED_INVSQRT                           FALSE
//    #define GMISC_INVSQRT_MIXED_ENDIAN               FALSE
//    #define GMISC_INVSQRT_REAL_SLOW                  FALSE
#define GMISC_NEED_MATRIXFLOAT2D                     TRUE
#define GMISC_NEED_MATRIXFIXED2D                     FALSE

#endif /* GFXCONF_H */
```

</details>

pointlessone, 2018-05-24:
It appears, ugfx doesn't pick up `gfxconfig.h` in my keymap dir.

Here's a fragment of expanded `lib/ugfx/src/gdisp/gdisp_image.c` that mentions `gfxconf.h`:

```
# 1 "lib/ugfx/src/gdisp/../../gfx.h" 1
# 42 "lib/ugfx/src/gdisp/../../gfx.h"
# 1 "keyboards/ergodox_infinity/gfxconf.h" 1
# 25 "keyboards/ergodox_infinity/gfxconf.h"
# 1 "quantum/visualizer/common_gfxconf.h" 1
# 26 "keyboards/ergodox_infinity/gfxconf.h" 2
# 43 "lib/ugfx/src/gdisp/../../gfx.h" 2
```

As you can see, it picks up keyboard's `gfxconf.h` instead of keymap's.

Here's the command for `gdisp_image.c` compilation:

```
arm-none-eabi-gcc -c -mcpu=cortex-m4 -fomit-frame-pointer -falign-functions=16 -ffunction-sections -fdata-sections -fno-common -fshort-wchar -DTHUMB_PRESENT -mno-thumb-interwork -DTHUMB_NO_INTERWORKING -mthumb -DTHUMB -ggdb -Os -Wall -Wstrict-prototypes -Werror -Wa,-adhlns=.build/obj_ergodox_infinity/lib/ugfx/src/gdisp/gdisp_image.lst -std=gnu99 -DCORTEX_VTOR_INIT=0x00002000 -DKEYBOARD_ergodox_infinity -DPROTOCOL_CHIBIOS -DCORTEX_USE_FPU=FALSE -include keyboards/ergodox_infinity/bootloader_defs.h -DLCD_WIDTH=128 -DLCD_HEIGHT=32 -DLED_WIDTH=7 -DLED_HEIGHT=7 -DGFX_USE_OS_CHIBIOS=TRUE -DGDISP_DRIVER_LIST=GDISPVMT_ST7565_QMK,GDISPVMT_IS31FL3731C_QMK -I./lib/chibios/os/license -I./lib/chibios/os/common/startup/ARMCMx/compilers/GCC -I./lib/chibios-contrib/os/common/startup/ARMCMx/devices/K20x -I./lib/chibios/os/common/ext/CMSIS/include -I./lib/chibios-contrib/os/common/ext/CMSIS/KINETIS -I./lib/chibios-contrib/os/common/startup/ARMCMx/devices/K20x7 -I./lib/chibios/os/rt/include -I./lib/chibios/os/common/oslib/include -I./lib/chibios/os/common/ports/ARMCMx -I./lib/chibios/os/common/ports/ARMCMx/compilers/GCC -I./lib/chibios/os/hal/osal/rt -I./lib/chibios/os/hal/include -I./lib/chibios/os/hal/ports/common/ARMCMx -I./lib/chibios-contrib/os/hal/ports/KINETIS/K20x -I./lib/chibios-contrib/os/hal/ports/KINETIS/LLD -I./lib/chibios-contrib/os/hal/boards/PJRC_TEENSY_3_1 -I./lib/chibios/os/hal/lib/streams -I./lib/chibios/os/various -Ikeyboards/ergodox_infinity -Idrivers/ugfx/gdisp/st7565 -Idrivers/ugfx/gdisp/is31fl3731c -I./lib/ugfx -I./lib/ugfx/3rdparty/tinygl-0.4-ugfx/include -Iquantum/visualizer -include keyboards/ergodox_infinity/config.h -MMD -MP -MF .build/obj_ergodox_infinity/lib/ugfx/src/gdisp/gdisp_image.td lib/ugfx/src/gdisp/gdisp_image.c -o .build/obj_ergodox_infinity/lib/ugfx/src/gdisp/gdisp_image.o
```
On closer inspection you can see that keybord is on the include search path (`-Ikeyboards/ergodox_infinity`) but the keymap (or user dir) is not.

drashna, 2019-11-21:
It's been a while since this was last updated. 

Also, we've been trying to move away from uGFX, due to a number of issues

closed 2019-11-21 by drashna