qmk/qmk_firmware #3091: RESET key not working with pro micro

Memnarch1113, 2018-05-31:
Hi all, I've been building a keyboard based on the ergodone (a split keyboard with a pro micro and an mcp23017). I was looking to put a software reset on it and subsequently put the `RESET` keycode onto a layer. When I activate that keycode, the keyboard goes unresponsive for a few seconds, then comes right back online like nothing happened (I have QMK toolbox running in the background to flash it when I reset the keyboard but it never says "Caterina device connected"). I noticed that [this issue](https://github.com/qmk/qmk_firmware/issues/164) seems to be similar, but it looks like it was fixed. Any help with the situation would be appreciated.

You can see my code [here](https://github.com/Memnarch1113/qmk_firmware). My code is in the ergodone folder (Haven't refactored it yet). Thanks for the help!

drashna, 2018-05-31:
https://github.com/Memnarch1113/qmk_firmware/blob/master/keyboards/ergodone/rules.mk#L68

Change this line to `BOOTLOADER = caterina`, recompile it and see if that fixes it.

100% certain that it will.

Memnarch1113, 2018-06-04:
Thanks very much. Haven't tried it, but that should absolutely do it.

closed 2018-06-04 by Memnarch1113

SkyCoops, 2019-09-01:
It worked, thank you very much.

It saved me from "bridging" the "reset" and "ground" pins on my ProMicro.