qmk/qmk_firmware #3131: Issue compiling firmware "sh: awk: command not found"

zjquid, 2018-06-06:
I'm quit new to this, so I'm sorry if I'm missing something obvious
I assume I forgot to install something... but I followed all of the setup steps (multiple times) and this issue keeps popping up

`Andrew@Andrew-PC MINGW64 /d/keyboards/qmk_firmware
$ make dz60:default
QMK Firmware 0.6.47
Making dz60 with keymap default

avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.5.4_1709) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

sh: awk: command not found
make[1]: *** [tmk_core/rules.mk:358: .build/obj_dz60_default/keyboards/dz60/dz60.o] Error 127
Make finished with errors
make: *** [Makefile:549: dz60:default] Error 1`

thanks for any help

zjquid, 2018-06-06:
thanks for a quick reply!
I tried that, but sadly still have the same result.

when I first built the board (dz60), I tried flashing the firmware with atmel flip like the Alps64, and it didn't work (probably an obvious mistake) but now the dz60 doesn't connect to the computer; I plug it in and I cannot type, and the LEDs on the bottom do not light up. (I assume this is because atmel flip flashed an invalid .hex file to the board, so the dz60 has firmware that doesn't match its physical components). I've been using a storebought mech keyboard (non-programmable) to do these steps in msys2 instead of using the dz60 I plan on programming with qmk...could this be causing any problems?

drashna, 2018-06-06:
This is two unrelated things. 
I'm not sure why MSYS2 is having issues for you.   Maybe run `pacman -Syuu` and see if that helps? 

As for the other, that's possible.  You can find the default firmware here: https://qmk.fm/keyboards/

Also, worst case, you can create the firmware at http://config.qmk.fm

zjquid, 2018-06-06:
I couldn't get the msys2 to work, however, building the firmware on config.qmk.fm and flashing the board with qmk toolkit worked like a charm!

That being said, what benefit is there in going the route of building firmware through msys2? It seems like using qmk toolkit is 500x easier...whatever

thanks so much!

drashna, 2018-06-06:
There are a lot of advantages. 
RIght now, you use the default config options for the board and firmware.  So if the board doesn't support RGB, no RGB. 
Custom macros are not supported.
The dual value keycodes don't work (such as LT, MT, etc) 
custom actions, such are RGB for layer indication are not possible.

So ... it really depends on what you want.  For most, the configurator should be ideal, mostly (or will be in the near future)

closed 2018-10-22 by drashna