qmk/qmk_firmware #3151: avr/pgmspace.h not found in nix-shell

pletcher, 2018-06-08:
Stack trace:

```
Compiling: keyboards/launchpad/launchpad.c
In file included from keyboards/launchpad/rev1/rev1.h:7:0,
                 from keyboards/launchpad/launchpad.h:5,
                 from keyboards/launchpad/launchpad.c:1:
./quantum/quantum.h:20:10: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
```

Steps to reproduce:

1. Run `nix-shell`
2. Run `sudo make launchpad:default`

I must be doing something wrong, right? It feels weird to use `sudo` in step 2.), but I get permissions errors about the files in `./quantum/` if I don't run the command as root.

OS: openSUSE Leap 15.0 (but this shouldn't matter with `nix-shell`, right?)

Thanks!

drashna, 2018-06-08:
rerun the install script.  It looks like the avr stuff didn't download right.

Namely run `util/linux_install.sh`, and see if that fixes the issue.

pletcher, 2018-06-11:
Hm, this is interesting:

```
⇒  ./util/linux_install.sh 
Sorry, we don't recognize your OS. Help us by contributing support!

    https://docs.qmk.fm/contributing.html
```

I had only run `util/install_dependencies.sh` before---it kind of worked, but it failed on a few packages. I just got those manually through zypper, but this error from running `util/linux_install.sh` directly makes a lot of sense.

Where should I start digging to add openSUSE support? (In the meantime, it looks like using Vagrant will allow me to build and flash just fine.)

drashna, 2018-06-11:
Honestly, I'm not sure how much I can help here, as I'm not "a linux guy". 

But it looks like you'd want to run both scripts, actually. 

Also, https://github.com/qmk/qmk_firmware/blob/master/util/linux_install.sh

You should read that.  And if you have a test system.... it may be worth testing this out until you get it "just right"

pletcher, 2018-06-11:
No worries, and thanks for the pointer! I'll poke around a bit and see if I can get something working.

drashna, 2018-06-12:
Oh, and if it helps: https://docs.qmk.fm/#/getting_started_build_tools?id=linux

Basically, if you get the list of those packages, and mapped correctly....

isolatedvirus, 2018-10-10:
i just ran into the same issue on my laptop (Tumbleweed, not leap).
Go into YaST and locate the avr-libc package, click on the package contents to find a file list, and add the location to /etc/profile (likely /etc/profile.local if Leap is similar to Tumbleweed)

For me, avr wasnt installed in the default location, and I had to add:
`export C_INCLUDE_PATH="/usr/avr/sys-root/include:${C_INCLUDE_PATH}"`

closed 2019-11-21 by drashna