qmk/qmk_firmware #3187: Changing SS Pin for SPI Bluetooth Communication

Suzaken, 2018-06-15:
The bluetooth code that Wez made has the SPI protocol communicate using the Slave select as the B0 pin on the atmega32u4. Is there a way to change the Slave Select pin to a different I/O pin? This is the code. [https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/lufa/adafruit_ble.cpp](url)

drashna, 2018-06-15:
Not sure if this would work (probably), 
Add something like this to the top:
```c
#ifndef AdafruitBleSSPin
#define AdafruitBleSSPin B0
#endif
```

And replace the call with: 
```c
    digitalWrite(AdafruitBleSSPin, PinLevelHigh);
    pinMode(AdafruitBleSSPin, PinDirectionOutput);
```

That should all it to work "normally" for the boards that use it, and should allow you to redefine the pin, if/when needed.

Suzaken, 2018-06-15:
Thanks for the help. Tried that, I feel like I'm still missing something. I believe that this part of the code would also have to be changed for me to redefine the pin, but I'm not sure what I would change. 
[https://github.com/qmk/qmk_firmware/blob/master/quantum/pincontrol.h](url)

drashna, 2018-06-16:
Unfortunately, that's about the extent of my expertise here.  Sorry.

fauxpark, 2020-03-17:
@Suzaken so the thing with the SS pin is it's required to initially be logic high for the AVRs to stay in SPI master mode, regardless of whether you're actually using it - B0 is left floating in the Feather schematic. The real slave select pin is in fact `AdafruitBleCSPin`.

PR #8299 (2020-04-08): spi_master for AVR

## Description

Pulling the SPI code up out of the BLE stuff so we can reuse it. Also replaced as much of the AVR-specific code as I could see, to facilitate Arm support by simply dropping in an appropriate spi_master.

Feel free to point out any dumb mistakes.

## Types of Changes

- [x] Core
- [ ] Bugfix
- [x] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* fixes #3187

## Checklist

- [x] My code follows the code style of this pro [... cut]
closed 2020-04-08 by tzarc