qmk/qmk_firmware #3188: How to avoid MODS state transfer from layer to layer (when pressed while TO() )

K-rnivoro, 2018-06-15:
I have a physical button to switch between to different layers suing TO(PC) and TO(VICE):

`
PC] = KEYMAP(
        KC_1,    KC_ESC,   CTL_T(KC_TAB), KC_LALT, KC_SPC,  KC_LGUI,    KC_Q,    TECLA_2,                                                                                                 \
        KC_3,    KC_W,     KC_A,          KC_LSFT, KC_Z,    KC_S,       KC_E,    KC_4,                                                                                                    \
        KC_5,    KC_R,     KC_D,          KC_X,    KC_C,    KC_F,       KC_T,    TECLA_6,                                                                                                 \
		TECLA_7, KC_Y,     KC_G,          KC_V,    KC_B,    KC_H,       KC_U,    TECLA_8,                                                                                                 \
		TECLA_9, KC_I,     KC_J,          KC_N,    KC_M,    KC_K,       KC_O,    TECLA_0,                                                                                                 \
		KC_PLUS, KC_P,     KC_L,          KC_COMM, KC_DOT,  TECLA_ACOR, KC_AT,   KC_MINS,                                                                                                 \
		KC_BSLS, KC_ASTR,  TECLA_CCOR,    KC_SLSH, MO(RST), KC_EQL,     KC_UP,   KC_HOME,                                                                                                 \
		KC_DEL,  KC_ENT,   KC_RGHT,       KC_DOWN, KC_F1,   KC_F3,      KC_F5,   KC_F7,                                                                                                   \
		                                                                                   KC_BSPC,                                                                                       \
																					                TO(VICE),                                                                             \
																					                            KC_P0,  KC_P8,  KC_P2,  KC_P4, KC_P6,                                    \
																										        KC_PDOT,  KC_P7,  KC_P3,  KC_P1,  KC_P9),	
// Emulator Mode. VICE (Using Positional layout) or CCS64
[VICE] = KEYMAP(
        KC_1,     KC_GRV,   KC_TAB,       KC_ESC,  KC_SPC,  KC_LCTRL,   KC_Q,    KC_2,                                   \
        KC_3,     KC_W,     KC_A,         KC_LSFT, KC_Z,    KC_S,       KC_E,    KC_4,                                   \
        KC_5,     KC_R,     KC_D,         KC_X,    KC_C,    KC_F,       KC_T,    KC_6,                                   \
		KC_7,     KC_Y,     KC_G,         KC_V,    KC_B,    KC_H,       KC_U,    KC_8,                                   \
		KC_9,     KC_I,     KC_J,         KC_N,    KC_M,    KC_K,       KC_O,    KC_0,                                   \
		KC_MINS,  KC_P,     KC_L,         KC_COMM, KC_DOT,  KC_SCLN,    KC_LBRC, KC_EQL,                                 \
		KC_INS,   KC_RBRC,  KC_QUOT,      KC_SLSH, KC_RSFT, KC_BSLS,    KC_DEL,  KC_HOME,                                \
		KC_BSPC,  KC_ENT,   KC_RGHT,      KC_DOWN, KC_F1,   KC_F3,      KC_F5,   KC_F7,                                  \
                                                                                           KC_PGUP,                      \
													                                                 TO(PC),             \
                                                                                                                 KC_P0,  KC_P8,  KC_P2,  KC_P4, KC_P6,	\
                                                                                                                 KC_PDOT,  KC_P7,  KC_P3,  KC_P1,  KC_P9),	
`
Thing is, if a press and hold ALT or CTRL, and switch between the two layers, the MODS keep pressed even if a release the key. I have to get back to previous layer and press it again to release it.
Funny thing: SHIFT works ok though...
There is a way to clear all the MODS when switching among layers?

K-rnivoro, 2018-06-15:
I think I have a clue: The layers have differente layout for ALT and CTRL keys, that's why those kepp pressed. when I switch between them...So, my question remains almost the same: Is there a way to clear the mods state regardless if they are pressed or not? How to put that in the KEYMAP sentence?

K-rnivoro, 2018-06-15:
Well Drashna...your ARE a genious...I've had this bug in my keyboard for a long time, and you fix it in a glance...

K-rnivoro, 2018-06-15:
I've read it. Now I want to add Mouse functionality to my Commmodore 64 keyboard!!! (officialy "Keymmodore 64").
Anyway, Thank you very much for your help. Not only fast but precise...

closed 2018-06-15 by K-rnivoro

drashna, 2018-06-15:
you're very welcome! 

Do you mean mouse keys? https://docs.qmk.fm/#/feature_mouse_keys

Or actual mouse support?  If this, check out the dichotomy keyboard, as that has it. :)

K-rnivoro, 2018-06-17:
Yes! I was talking about mouse_keys, as a "spare" mouse when you cannot
find one.
BTW, You remember my project? A Commodore 64 keyboard (and it's joysticks)
to USB.
With your help, NOW is finished. Has a lot of features thanks to qmk like
CTRL key being TAB if tapped, and CTRL if held
(remember where the CTRL key is located).
Also an RGB led indicating which layer is active (where the PWR led was).
I want to share it but I need help to put everything in order.
What do you think? Is it worth it?

K-rnivoro, 2018-06-18:
> Awesome!
> And yes, definitely share it!

Ok but I Think I'll need help with the documentation and stuff. I've made
several oprtions and I do not know how to organize all the info in a
"clean" format.

I can show you my stuff first, if you promess not to distribute it.