qmk/qmk_firmware #3192: Ctrl -> Space Cadet Shift quick taps sends Ctrl+Shift+9

ArcticZeroo, 2018-06-16:
When you tap Ctrl, then tap Space Cadet Shift (open/left), it sends Ctrl+Shift+9 instead of just Ctrl+Shift or Ctrl+( or Ctrl+Shift+( (idunno what intended behavior _should_be, but I'm sure this is not it?)

The only reason I noticed this is because I have Ctrl+Shift+9 bound to screen recording in ShareX and it makes for a lot of accidental recordings (apparently I hit ctrl a lot while I'm shifting)

drashna, 2018-06-16:
Unfortunately, it sounds like this is "normal behavior" for the space cadet code. :(

https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L554-L596

closed 2018-10-22 by drashna

skorokithakis, 2020-03-09:
@drashna Can this be reopened? I hit this a lot when using Alt+Shift and other keys, and I believe a way to tell Space Cadet that its purpose has been fulfilled (and that it doesn't need to emit the tap key) is sorely needed.

Alternatively, would we be able to emulate this better by using Tap Dance? My usage is:

* Space Cadet shifts (pressing is parenthesis, holding is shift).
* Alt+Shift passing through as normal (no parentheses, no matter whether I instantly tap Shift).
* Left shift+Right shift sending "()", as sometimes I haven't managed to release left shift before tapping right.
* Right shift+Left shift sending Caps Lock.

I currently can't make these work, because if I tap a Shift key quickly, its tapping keycode is always emitted, even if it was part of Alt+Shift for changing the language, and there's no way to change this from my config because I can't tell Space Cadet to not emit the parenthesis no matter how short the tap is.