qmk/qmk_firmware #3211: When flashing QMK DFU onto Pro Micro many of the pins seem to not work

drashna, 2018-06-21:
To clarify, the bootloader works fine, and the pin config is correct (can escape properly). 

But once it's in the firmware, a bunch of the keys stop working.  like the pins are wrong. 

For anyone that wants to try, the fuses are listed here: 
https://github.com/qmk/qmk_firmware/blob/master/users/drashna/readme.md#pro-micro-hacking
But are: Low: 0x5E High: 0x99 Extended: 0xF3 Lock: 0xFF
(defaults are: Low: 0xFF High: 0xD8 Extended: 0xC3 Lock: 0x3F)

On my viterbi (left hand), columns 3-6 (starting at 0) work fine. But columns 0-2 are unresponsive.  All rows appear to work. 

Viterbi pin config is here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/viterbi/rev1/config.h#L37-L38

drashna, 2018-06-21:
So, it looks like it's the F5, F6, F7 pins that the ones that are affected.

fauxpark, 2018-06-21:
Sounds like JTAG is enabled - it turns on the internal pullups for F4, F5 and F7 (and F6 is tristated). Change the high fuse to 0xD9 or see below.

QMK [does make sure to turn it off for certain MCUs](https://github.com/qmk/qmk_firmware/blob/master/quantum/matrix.c#L151) in `matrix_init` - but I see this keyboard uses its own `matrix.c` which doesn't have that code.

drashna, 2018-06-21:
Will try both sortly.

closed 2018-06-22 by drashna

mcmillion, 2018-11-02:
I think I'm having the same issue with my new Viterbi build with an Elite-C. Depending on which side I plug the USB into, the opposite side's final 3 columns don't respond. What was the fix here?