qmk/qmk_firmware #3246: Compile error when trying to compile Marlin sketch for Arduino Mega 2560

bfrakes256, 2018-06-28:
Here is the exact error:

Archiving built core (caching) in: C:\Users\<username>\AppData\Local\Temp\arduino_cache_366030\core\core_arduino_avr_mega_cpu_atmega2560_cc708c5a95af27b60b589ff2ba91974d.a
fork/exec C:\Users\<username>\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

I've updated all boards and libraries in the Arduino IDE

Marlin version Bugfix2.0.x

Please advise....

drashna, 2018-06-28:
I think you have the wrong repo here?   
This is for QMK, the keyboard firmware.   

And it uses a downloaded avr-gcc version, and do not use the Arduino IDE.  (it did use winAVR a long time ago, but isn't recommended now). 

You can find out how to set up the build environment here: https://docs.qmk.fm/#/newbs

bfrakes256, 2018-06-28:
Yes, I've put the same post on the Marlin forum.  Sorry for the goof...

drashna, 2018-06-28:
Not a problem. :D

closed 2018-06-28 by drashna