qmk/qmk_firmware #3250: using fn1 key as ctrl when not pressed with any key, but when it is, it is used as function key

Jungwoo0213, 2018-06-28:
Hi, I am very new to QMK and trying to move from easyAVR

I am using GH60 Satan.

Currently my Capslock key is mapped as fn1, so that I can access arrow keys, home, end etc.
What I am trying to achieve is Capslock(fn1) work as left Ctrl key when it is held down without pressing any other key. But if i press other keys with fn1 held down, I can access other keys in the layer1.

I want the ctrl to be in the place of Capslock because I like to browse the internet with the ctrl held down, because ctrl+mouse click opens up the link in the new window.

Thanks for your help!

henrebotha, 2018-06-28:
That can't really work unless you do it with a timeout, otherwise how is the key supposed to know whether you're going to press other keys or not?

Jungwoo0213, 2018-06-28:
What do you mean by timeout? is that same is layer/toggle?
I thought it might be possible after noticing that you can write conditional statements in c code.

henrebotha, 2018-06-28:
Ok think about it like this. I'm going to assume you use <kbd>Fn</kbd><kbd>W</kbd> for <kbd>↑</kbd>.

# Scenario 1

1. At t=0s, you press <kbd>CapsLock</kbd> down.
1. Two seconds pass.
1. At t=2s, you press <kbd>W</kbd> down. At this point, QMK realises you are trying to do <kbd>Fn</kbd><kbd>W</kbd>, so it outputs <kbd>↑</kbd>.

# Scenario 2

1. At t=0s, you press <kbd>CapsLock</kbd> down.
1. Two seconds pass.
1. At t=2s, you haven't pressed anything yet, so QMK can't output anything yet.

Now, what is the difference between scenario 1 and scenario 2 at t=1s? There is no difference. In both scenarios, from QMK's perspective, you have pressed down a key and have not done anything else yet.

So how is it supposed to know when to emit <kbd>Ctrl</kbd>?

Jungwoo0213, 2018-06-28:
For ctrl+mouse left click you don't have to release ctrl. I can click while Ctrl is down as many times I want.

my Scenario is that

Scenario 1 (trying to ctrl + mouse click)
1. press Capslock down --> Triggers ctrl down
2.  the keyboard does not sense any keypress
3. When user releases Capslock --> triggers ctrl up

Scenario 2 (trying to use Fn +W)
1. press Capslock down --> triggers ctrl down
2. the keyboard senses that I pressed W
3. release ctrl and trigger fn1 changing to layer1
4. register W
5. release fn1

My solution is to emit ctrl anyway.

Snipeye, 2018-06-28:
I have implementation something along these lines in my keyboard: I have ENTER and APOSTROPHE mapped to the same key, and which one is emitted depends on how long I hold it. If I press the key down, a timer is started.  When the timer reaches the threshold, enter is registered.  When the key is released, enter is unregistered.  If the timer does not reach the threshold and the key is released, a macro writes an apostrophe via SS_TAP (I think that's what I used, anyway).

This could be adapted to your situation: key is pressed, a bool is set.  If any other key is pressed, it assumes that your key was intended as "control" and registers control before it process the rest of the keypresses.  If your key is released, however, before any other keys are pressed, it would assume you wanted to change layers.

Or do I have it backward?

Jungwoo0213, 2018-06-30:
Thanks all for your help!

@Snipeye 
For the timer solution, I have tried using my space to key to act as ctrl (pressed long, act as ctrl), it just is not as quick as before. I have to give some delay before I press anything with the mouse, or it will register as space. Although this was quite a solution, I found this kind of slowing down my work flow.

For now, I think I will go with shift + mouse left click using autohotkey.

Again,  thank you guys for help.

ps. for the fn solution, I think I will study macro to find the way out....

closed 2018-10-22 by drashna