qmk/qmk_firmware #3251: Build error(s): implicit declaration of function 'KEYMAP' / initializer element is not constant

henrebotha, 2018-06-28:
System:

* macOS 10.13.5
* Zsh 5.5.1
* GNU Make 3.81

Build errors:

```
QMK Firmware 0.6.49
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making ergodox_ez with keymap henrebotha

avr-gcc (GCC) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_ez/twimaster.c                                                         [OK]
Compiling: keyboards/ergodox_ez/matrix.c                                                            [OK]
Compiling: keyboards/ergodox_ez/ergodox_ez.c                                                        [OK]
Compiling: keyboards/ergodox_ez/keymaps/henrebotha/keymap.c                                        keyboards/ergodox_ez/keymaps/henrebotha/keymap.c:33:18: error: implicit declaration of function 'KEYMAP'; did you mean 'KEYMAP_H'? [-Werror=implicit-function-declaration]
   [LAYER_HOME] = KEYMAP(
                  ^~~~~~
                  KEYMAP_H
keyboards/ergodox_ez/keymaps/henrebotha/keymap.c:33:18: error: initializer element is not constant
keyboards/ergodox_ez/keymaps/henrebotha/keymap.c:33:18: note: (near initialization for 'keymaps[0][0][0]')
```

It's been a while since I've built my firmware, so I don't know whether something fundamental has changed in QMK in the meantime. My keymap is a fork of the Ergodox EZ default keymap.

henrebotha, 2018-06-28:
In case it's relevant, my keymap lives here: henrebotha/qmk_keymap

drashna, 2018-06-28:
The issue is that avr-gcc 8.1.0 isn't fully compatible with our build environment. 
To fix this issue, you'll need to revert the version.  To do so, run:
```
brew install avr-gcc@7.3
brew pin osx-cross/avr/avr-gcc
```
This will revert to a working version, and prevent it from updating.

henrebotha, 2018-06-28:
Sick, thanks @drashna!

henrebotha, 2018-06-28:
Hang on... This doesn't work, I get the same issues with 7.3.0.

```
QMK Firmware 0.6.49
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making ergodox_ez with keymap henrebotha

avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_ez/twimaster.c                                                         [OK]
Compiling: keyboards/ergodox_ez/matrix.c                                                            [OK]
Compiling: keyboards/ergodox_ez/ergodox_ez.c                                                        [OK]
Compiling: keyboards/ergodox_ez/keymaps/henrebotha/keymap.c                                        keyboards/ergodox_ez/keymaps/henrebotha/keymap.c:33:18: error: implicit declaration of function 'KEYMAP'; did you mean 'KEYMAP_H'? [-Werror=implicit-function-declaration]
   [LAYER_HOME] = KEYMAP(
                  ^~~~~~
                  KEYMAP_H
keyboards/ergodox_ez/keymaps/henrebotha/keymap.c:33:18: error: initializer element is not constant
keyboards/ergodox_ez/keymaps/henrebotha/keymap.c:33:18: note: (near initialization for 'keymaps[0][0][0]')
```

drashna, 2018-06-28:
Ah, sorry, `KEYMAP` should be `LAYOUT_ergodox`

henrebotha, 2018-06-28:
Alright, seems to compile with that change. Thanks. 

Do I understand correctly that the `_pretty` keymaps simply reorder the keys so you can lay out your keymap side-by-side in code?

closed 2018-06-28 by henrebotha

drashna, 2018-06-28:
Yes, correct. It's more "visually correct", but it's just a reordering of the matrix to do so.