qmk/qmk_firmware #3286: Verification Error when Flashing Bootloader

PickleJr, 2018-07-02:
I'm at a complete loss here on what to do. I've ordered a new flasher in case the issue is with my flasher. But aside from that I'm not sure what my issue could be. The error in question is this:
```
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x1e000 0xff != 0x0c
avrdude: verification error; content mismatch
avrdude: safemode: Fuses OK (E:F3, H:99, L:5E)
```
I'm using a Sparkfun Pocket AVI, and I'm trying to flash the planck light. I know I'm able to connect to the board. Running `avrdude -p usb1287 -c usbtiny` confirms I have good contact with the board. The commands I have ran are these:
`sudo avrdude -p usb1286 -c usbtiny -U flash:w:bootloader_at90usb128x_1_0_1.hex:i`
and
`sudo avrdude -p usb1287 -c usbtiny -U flash:w:bootloader_at90usb128x_1_0_1.hex:i`
(the one difference is the usb128**7** and usb128**6**. I'm a little bit of a noob so I'm not sure if there's much difference in the commands.)
I have tried running this: `sudo avrdude -p usb1286 -c avrispmkII -U flash:r:mystery.hex:r` to see if the bootloader is getting flashed at all. When I run this command the file that gets returned is blank. There's nothing in it. No bootloader. So I'm not sure if I'm even writing to the bootloader, or if I'm unable to read it.
On my pocket avr, the connections I have set up are:
```
RST <-> RST
GND <-> GND
B1  <-> SCK
B2  <-> MIS1
B3  <-> MIS0
VCC <-> VCC
```
I have tried on both a Windows and a Linux laptop. So I don't think the issue is my laptop setup. My Linux computer is being a little weird I think? When I run the flash command it "writes" and "reads" kind of fast before it verifies (progress bars jump to 100%.) whereas my windows laptop seems to go at a slower speed (I can see the progress bar creep up to 100%).
The planck did come in working order. It lit up, buttons worked, and everything. After the first flash to install the correct bootloader however, it never woke up. I understand it's expected the lights wont work, but the computer wont even read keystrokes. I'm hoping it's not actually fried somehow. I do think there's life however. When I plug it in windows makes a "device connected" sound. I just can't do anything with it.
I know the issue isn't with the USB cable connecting the laptop to the programmer either. I've tried multiple cables. 
Thanks for taking your time at reading this.

drashna, 2018-07-02:
It sounds like the fuses are not set correctly, and that may be causing the mismatch error.

And for the Planck Light, it looks like it should be the AT90USB128**6**: https://github.com/qmk/qmk_firmware/blob/master/keyboards/planck/light/rules.mk#L5

PickleJr, 2018-07-02:
Is my wire connection wrong? Wouldn't `avrdude -p usb1286 -c usbtiny` throw a fit if the wires were misconfigured? I could try a different set of jumper wires but I doubt that's where the issue lies since I've tried two different sets. Or am I misunderstanding what you mean by the fuses not being set correctly?

drashna, 2018-07-02:
On windows, try using avrdudess.  It's a GUI and may be a bit easier to use. Also, try the "detect MCU" option.  If that comes back with a good result, then it's wired correctly.  (I don't know/remember the commands to test this).

But it looks like you do have the right pins.   However, I suspect that you don't have the right fuses set. But playing with those .... is a lot trickier and you could cause serious issues.

PickleJr, 2018-07-02:
Thanks for the tip on avrdudess. It looks like my board is hooked up alright. I found this page when it comes to setting up fuses with avrdudess [http://www.engbedded.com/fusecalc](http://www.engbedded.com/fusecalc) and.... you're right. Most of what's on this page is over my head. I'm not sure where to go to from here.

PickleJr, 2018-07-02:
Weird, if I flash another boot-loader, like the bootloader_atmega16u4_1_0_1.hex, the bootloader flashes fine. I'm able to confirm that it writes by calling the read command again. It just has a verification error on at90usb128x for some reason.

drashna, 2018-07-04:
In that case, it sounds like one of the fuses is not properly set.  

Unfortunately, the fuses are one of the things that you can cause a lot of issues with. And it's one of the areas that I don't understand very well. 

@skullydazed any suggestions?

PickleJr, 2018-07-11:
Update: Went to microcenter, got myself a teensy, and flashed the bootloader with that. I'm not sure why the sparkfun pocket avi didn't work, but everything is good now. I typed this up on my planck. Thank you so much for taking the time to reply and help though!

drashna, 2018-07-20:
Awesome, glad to hear it!

closed 2018-07-20 by drashna

sigprof, 2022-01-16:
USBtinyISP and any other ISP flashers using the same `usbtiny` protocol and firmware (e.g., SparkFun Pocket AVR Programmer) **DO NOT** support AVR chips with more than 64 KiB of flash, therefore they cannot be used to flash the AT90USB1286/7 chips.

At the moment the [shop page for SparkFun Pocket AVR Programmer](https://www.sparkfun.com/products/9825) mentions this in the description:

>  The microcontroller-to-be-programmed can be any AVR with 64K or less of flash.

Adafruit has the same information in the [FAQ for USBtinyISP](https://learn.adafruit.com/usbtinyisp/f-a-q#faq-2270879):

> Any AVR that uses the ISP interface for programming and has 64K or less of flash can be programmed.
> 
> Chips such as the Atmega1280/1281 and Atmega2560/2561 have more than 64K and cannot be programmed.

The limitation is due to the usage of 16-bit addresses in the `usbtiny` protocol, therefore it cannot be fixed without updating not just the firmware in the flasher (for which you would need to have a second ISP flasher anyway), but also the protocol implementation in `avrdude` (or completely rewriting the firmware to use a different communication protocol with support for more address bits).