qmk/qmk_firmware #3302: QMK DFU exit not working correctly?

kdb424, 2018-07-04:
I have a levinson R2 and have been trying for a few days to get my bootloader to exit when hitting the escape key (Top left key of left board). I have refrenced this for pinout as that's the code that I am using on the board being configured. 
https://github.com/qmk/qmk_firmware/blob/master/keyboards/levinson/rev2/config.h
I am flashing with DFU directly to the board as it's already installed. Is this something that I should have correctly set when initially installing or will a line like this not work?

make levinson/rev2:kdb424:dfu

It will of course flash everything else correctly (layout, and rules changes)

drashna, 2018-07-04:
Try all the keys, just in case?
And try reversing these?

kdb424, 2018-07-04:
Tried every key on both boards, and have reversed them. Still no luck unfortunately.

drashna, 2018-07-05:
dang.  Not sure then.

fauxpark, 2018-07-05:
My guess: JTAG again. The QMK bootloader [doesn't appear to disable it](https://github.com/qmk/qmk_firmware/blob/master/lib/lufa/Bootloaders/DFU/BootloaderDFU.c) except when `BOARD == BOARD_XPLAIN` (and even then it's re-enabled shortly afterwards).

drashna, 2018-07-06:
The problem with that theory, is that it doesn't apply to the 32u4.  
Also, I have this issue on my viterbi's, which has the fuse set to disable JTAG:
Low: 0x5E High: 0xD9 Extended: 0xC3 Lock: 0x3F

And I have the same issue.   It may still be JTAG related, but I'm not sure.

fauxpark, 2018-07-06:
Not sure what you mean by "it doesn't apply to the 32U4". I just recently encountered the issue on a board I'm designing, that I can't actually change the fuses or bootloader on due to a lack of ISP header.

drashna, 2018-07-06:
What I mean this: 
https://github.com/qmk/qmk_firmware/blob/master/lib/lufa/Bootloaders/DFU/BootloaderDFU.c#L111-L121

It looks like JTAG isn't messed with at all, on "BOARD_LEONARDO" which is the 32u4.

fauxpark, 2018-07-06:
`BOARD_LEONARDO` does imply the 32U4, but the 32U4 does not imply `BOARD_LEONARDO`. It's never used in any keyboard, I checked :) What you should be looking at is the code in the `#else` - which also doesn't disable JTAG - my point exactly.

drashna, 2018-07-06:
Okay, thanks for the clarification and dealing with my dumb ass. :)

Also, if JTAG is disabled via fuses, then this shouldn't be an issue, correct?

fauxpark, 2018-07-06:
Correct, that's what I can't figure out. Your fuses do look right.

fauxpark, 2018-07-06:
According to the datasheet it sounds like JTAG *cannot* be enabled in software when the `JTAGEN` fuse is unprogrammed... Hmm...

drashna, 2018-07-07:
I personally set the High fuse to 0xD9, which should disable JTAG, and that works fine for me (2 irises, 2 orthodox and 1 viterbi)

@kdb424 do you know what the fuse settings for your controller are?  You can use the ISP flasher to read them, if need be.

fauxpark, 2018-07-07:
Run this (assuming you're using the same programmer as in the link):
```
avrdude -c usbtiny -p m32u4 -P usb -U hfuse:r:-:i
```

That will write the high fuse to the console in the same format as a .hex file, but underneath it should also output the rest of the fuses in a more readable format. According to your link you should get:

```
avrdude: safemode: Fuses OK (E:C3, H:D9, L:5E)
```

kdb424, 2018-07-07:
To be clear, this is being done with a programmer soldered to the pro micro, not on the pro micro itself? I was using a teensy 2.0 when doing this. I'll have to break out the soldering iron as I forgot to install the programming headers on my board, so want to be sure.

fauxpark, 2018-07-07:
Yeah, the 1x6 header on the keyboard halves themselves should allow you to ISP program the Pro Micros now that they've been soldered on.

kdb424, 2018-07-07:
Well, should have remembered to put those pins on the header. I'll either fire up the soldering iron or borrow my friends Iris as we remembered to put the pins on his unless someone beats me to the punch.

closed 2019-11-21 by drashna

Torreip, 2020-04-22:
Maybe some pin configurations are not working?
I had it on output D2, input D3 and it was working fine with tweezers.
I moved it to output B6, input B5 and it was not working. I checked B6 with a multimeter, no signal.
I tried output B5, input B6 (inversing to check) and could read logical 1 at B5 (but because of the diode it could not work).
I tried output F5, input B5 and it is working.

So for some reason, output B6 does not work on my pro micro. I am happy with F5/B5 though.

Kosaro, 2021-04-22:
I'm having this problem on the Elite-C, but I'm a bit confused with what I need to do as DFU is already preinstalled.

kdb424, 2021-04-22:
@Kosaro This thread is about QMK's DFU fork, not the standard version that is installed on the Elite-C