qmk/qmk_firmware #3316: make planck/light:default doesn't compile (MIDI issue)

sanosis, 2018-07-05:
The default keymap with current settings don't compile.

`make planck/light:default`

Gives errors relating to MIDI, turning MIDI off in
keyboards/planck/light/rules.mk
from
`MIDI_ENABLE = yes`
to
`MIDI_ENABLE = not`
and compiling produces a hex file

drashna, 2018-07-05:
Compiles fine for me.

How are you compiling? 
Config.qmk.fm?  Or locally? And if so, on which OS?

And when you compile, what version does it say (eg: `QMK Firmware 0.6.57`)?

sanosis, 2018-07-05:
Arch Linux - more or less current. Git clone of repo.

drashna, 2018-07-05:
When compiling, does it say `AVR GCC 8.1.0` by chance?  And is it erroring out on LUFA stuff? 

If so, then you need to roll back to AVR GCC 7.3.0, until we're able to fix the issue.

sanosis, 2018-07-05:
Correct - avr-gcc (GCC) 8.1.0
QMK Firmware 0.6.57
exact errors::

Compiling: lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.c                               In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:380:5: error: 'const' attribute on function returning 'void' [-Werror=attributes]
     void Audio_Device_Event_Stub(void) ATTR_CONST;
     ^~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:382:10: error: 'EVENT_Audio_Device_StreamStartStop' alias between functions of incompatible types 'void(USB_ClassInfo_Audio_Device_t * const)' {aka 'void(struct <anonymous> * const)'} and 'void(void)' [-Werror=attribute-alias]
     void EVENT_Audio_Device_StreamStartStop(USB_ClassInfo_Audio_Device_t* const AudioInterfaceInfo)
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.c:191:6: note: aliased declaration here
 void Audio_Device_Event_Stub(void)
      ^~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:360: .build/obj_planck_light_default/./lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.o] Error 1
Make finished with errors
make: *** [Makefile:549: planck/light:default] Error 1

drashna, 2018-07-05:
Yup. That's the issue I thought it was.  

You'll need to revert back to the 7.3.0 version of AVR GCC to fix this, until there is a fix for QMK.

sanosis, 2018-07-06:
I'm using Arch Linux, but it's a tag irrelevant now. My Plank Light died on me, so won't have a way to test the bootloader/keymaps, and 7.3.0 works well enough with jj40 keymaps.

PR #3510 (2018-09-11): Backport fixes for GCC 8.1
Hi,

This PR just backports https://github.com/abcminiuser/lufa/commit/d8e0d67caefa312970787645180ec5db2eff92cb and https://github.com/abcminiuser/lufa/commit/a08a02481ba9e68933ad0b89483f5328767df9f3 so that QMK compiles with avr-gcc 8.1.

Before this PR, I would get:

```
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.c                               In file included from lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.c:38:
lib/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.h:382:10: error: 'EVENT_Audio_Device_StreamStartStop' alias  [... cut]
closed 2018-09-11 by skullydazed