qmk/qmk_firmware #3321: ShinyDox merge nuked the Ergodox EZ

drashna, 2018-07-05:
The shinydox merge breaks a bunch of stuff for the Ergodox board. 

First is that the RGB defaults to underglow off, and matrix on. 
Second, is that even when swapping these (Underglow on, and Matrix off), there is an issue with includes and it won't compile right (for instance, my code won't compile, unless I add an include for rgblight. 
Third .... not sure if it's fully related, but I can't use `ifdef KEYBOARD_ergodox_ez` or any other ifdef for my ergodox layout, as it will not detect it, for whatever reason.  
Fourth, even when getting RGB underglow working, it is displayed wrong. It no longer bit bangs the color to the left half, and the right half only shows part of the strip lit up... and it's random colors. 

So, the changes HARD broke the ergodox board. 

I can issue a PR to revert the changes introduced by the `hf/shinydox` branch, but I'd rather see the issue fixed.  (let me know, I'll definitely test!)

Tagging @jackhumbert @ezuk to let you guys know.

drashna, 2018-07-05:
As a stop gap, 'git revert 9c2dde98e2b963704eb7cb87f6c53c52599fba53 -m 1' will fix this issue locally, by reverting the change.

jackhumbert, 2018-07-09:
@drashna It looks like the breaking changes to the RGB stuff are [here](https://github.com/qmk/qmk_firmware/commit/9c2dde98e2b963704eb7cb87f6c53c52599fba53#diff-d9aeaf00f769c7dedfefaebfe43fece7R84).

Adding in the rgblight.h include somewhere and fixing the defaults for whatever @ezuk wants would make sense!

drashna, 2018-07-09:
You're right about the section. Specifically, it needs these still: 
```c
#define RGBW_BB_TWI
#define RGBW 1
```
It needs both, as well. I'll open a PR and tag ezuk in it. 

That said... the `ifdef KEYBOARD_ergodox_ez` in the make file DOES NOT work. It used to. But now, it doesn't.   And the problem is that I was wrapping the rgblight enable with that.  

Eg, this does not work:
```c
ifdef KEYBOARD_ergodox_ez
  RGB_MATRIX_ENABLE = no
  RGBLIGHT_ENABLE   = yes
endif
```
But this does:
```c 
  RGB_MATRIX_ENABLE = no
  RGBLIGHT_ENABLE   = yes
```

I'm not even sure why this would cause this change, as nothing in the code should

drashna, 2018-07-09:
.... 
Wow.
Just for documentation purposes,
```c 
ifdef KEYBOARD_ergodox_ez
  RGB_MATRIX_ENABLE = no
  RGBLIGHT_ENABLE   = yes
endif
```
does not work, but
```c
ifdef KEYBOARD_ergodox_ez
  RGBLIGHT_ENABLE   = yes
  RGB_MATRIX_ENABLE = no
endif
``` 
does work just fine. 

Makefiles (rules.mk) are still a lot of magic to me, so no idea why this matters (thouglh rgblight and then rgb matrix is the order that common_features lists these in, as well as how the ergodox's rules.mk lists them in too, so ... maybe that's why?). 

Either way, I can say that this is closed now.

closed 2018-07-09 by drashna

drashna, 2018-07-09:
Wrong.  I need to use `ifeq ("$(KEYBOARD)","ergodox_ez")`, not `ifdef KEYBOARD_ergodox_ez`

PR #3354 (2018-07-09): Fix Ergodox EZ issues introduced with ShinyDox merge (#3229)
This re-enables the RGBW and BitBang features for the RGB Underglow, as it breaks compiling if you want to enable RGB underglow rather than rgb matrix.

Also, defaults to RGB underglow, and NOT matrix (for now).

@ezuk This look good? 

Fixes #3321