qmk/qmk_firmware #3343: How to compile QMK for Ergodox EZ with Tap Dance enabled?

zzz2496, 2018-07-08:
Hi,

I've been struggling how to compile QMK for Ergodox EZ with Tap Dance enabled. 
I'm on Mac OS High Sierra 10.13.5, installed Git Client for Mac through https://git-scm.com/download/gui/mac and updated my Homebrew installation.

Here's what I've done for QMK:
1. Clone the repo using command: git clone https://github.com/qmk/qmk_firmware.git
2. I opened the qmk_firmware directory, and finish the installation using command: ./util/qmk_install.sh
3. Then I open my configuration: https://configure.ergodox-ez.com/keyboard_layouts/aggvwo/edit
4. Made a new folder "damir" under: qmk_firmware/keyboards/ergodox_ez/keymaps/
5. Download the source code, and put it in: qmk_firmware/keyboards/ergodox_ez/keymaps/damir then renamed the file to keymap.c
6. Edit the file, change every instance of "KEYMAP" function to "LAYOUT_ergodox" so that I can build the firmware, otherwise it will fail.
7. I edited qmk_firmware/keyboards/ergodox_ez/config.h file, changed the TAPPING_TERM from 200 to 175 (I need a slightly more responsive dual-function key).

Up to this stage, I got a fully operational firmware BUT WITHOUT Shine LED control. As I understand, there are changes in Ergodox EZ that is breaking the firmware if I understand correctly - it builds, but the Shine LED control is not functioning.

Continuing:
7. I edit qmk_firmware/keyboards/ergodox_ez/rules.mk file, added TAP_DANCE_ENABLE = yes
8. I build the firmware and got this result: 
```
QMK Firmware 0.6.57
Making ergodox_ez with keymap damir

tmk_core/rules.mk:354: target `.build/obj_ergodox_ez_damir/i2c_master.d' given more than once in the same rule.
avr-gcc (GCC) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  28824	      0	  28824	   7098	./.build/ergodox_ez_damir.hex

Compiling: keyboards/ergodox_ez/keymaps/damir/keymap.c                                              [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/ergodox_ez_damir.elf                                                                [ERRORS]
 |
 | .build/obj_ergodox_ez_damir/quantum/process_keycode/process_tap_dance.o: In function `process_tap_dance_action_on_reset':
 | /Users/koko/qmk_firmware/quantum/process_keycode/process_tap_dance.c:107: undefined reference to `tap_dance_actions'
 | /Users/koko/qmk_firmware/quantum/process_keycode/process_tap_dance.c:107: undefined reference to `tap_dance_actions'
 | .build/obj_ergodox_ez_damir/quantum/process_keycode/process_tap_dance.o: In function `_process_tap_dance_action_fn':
 | /Users/koko/qmk_firmware/quantum/process_keycode/process_tap_dance.c:85: undefined reference to `tap_dance_actions'
 | /Users/koko/qmk_firmware/quantum/process_keycode/process_tap_dance.c:85: undefined reference to `tap_dance_actions'
 | .build/obj_ergodox_ez_damir/quantum/process_keycode/process_tap_dance.o: In function `process_tap_dance_action_on_reset':
 | /Users/koko/qmk_firmware/quantum/process_keycode/process_tap_dance.c:108: undefined reference to `tap_dance_actions'
 | .build/obj_ergodox_ez_damir/quantum/process_keycode/process_tap_dance.o:/Users/koko/qmk_firmware/quantum/process_keycode/process_tap_dance.c:108: more undefined references to `tap_dance_actions' follow
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/ergodox_ez_damir.elf] Error 1
make: *** [ergodox_ez:damir] Error 1
Make finished with errors
```
This is where I hit the dead end. I have no idea what it's saying, why does it error as such. If I understand correctly, setting [TAP_DANCE_ENABLE = yes] should not break anything, but I might be wrong. I've tried putting this code into my keymap.c file:
```
enum {
  TD_ESC_CAPS = 0
};

//Tap Dance Definitions
qk_tap_dance_action_t tap_dance_actions[] = {
  //Tap once for Esc, twice for Caps Lock
  [TD_ESC_CAPS]  = ACTION_TAP_DANCE_DOUBLE(KC_ESC, KC_CAPS)
// Other declarations would go here, separated by commas, if you have them
};

//In Layer declaration, add tap dance item in place of a key code
TD(TD_ESC_CAPS)

```
The result: 
```
QMK Firmware 0.6.57
Making ergodox_ez with keymap damir

tmk_core/rules.mk:354: target `.build/obj_ergodox_ez_damir/i2c_master.d' given more than once in the same rule.
avr-gcc (GCC) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  28824	      0	  28824	   7098	./.build/ergodox_ez_damir.hex

Compiling: keyboards/ergodox_ez/keymaps/damir/keymap.c                                             In file included from ./quantum/quantum.h:100,
                 from keyboards/ergodox_ez/ergodox_ez.h:4,
                 from keyboards/ergodox_ez/keymaps/damir/keymap.c:1:
./quantum/process_keycode/process_tap_dance.h:36:29: error: expected ')' before '+' token
 #define TD(n) (QK_TAP_DANCE + n)
                             ^
keyboards/ergodox_ez/keymaps/damir/keymap.c:23:1: note: in expansion of macro 'TD'
 TD(TD_ESC_CAPS)
 ^~
In file included from keyboards/ergodox_ez/keymaps/damir/keymap.c:1:
keyboards/ergodox_ez/keymaps/damir/keymap.c:37:690: error: 'RGB_SLD' undeclared here (not in a function); did you mean 'RGB_SPD'?
   [1] = LAYOUT_ergodox(KC_ESCAPE,KC_F1,KC_F2,KC_F3,KC_F4,KC_F5,KC_F6,KC_TRANSPARENT,KC_EXLM,KC_AT,KC_LCBR,KC_RCBR,KC_PIPE,KC_TRANSPARENT,KC_TRANSPARENT,KC_HASH,KC_DLR,KC_LPRN,KC_RPRN,KC_GRAVE,SFT_T(KC_NO),KC_PERC,KC_CIRC,KC_LBRACKET,KC_RBRACKET,KC_TILD,KC_TRANSPARENT,KC_TRANSPARENT,KC_TRANSPARENT,KC_AUDIO_MUTE,KC_AUDIO_VOL_DOWN,KC_AUDIO_VOL_UP,KC_LEFT,KC_RIGHT,KC_TRANSPARENT,RGB_VAD,RGB_VAI,KC_TRANSPARENT,KC_F7,KC_F8,KC_F9,KC_F10,KC_F11,KC_F12,KC_BSPACE,KC_TRANSPARENT,KC_UP,KC_7,KC_8,KC_9,KC_KP_ASTERISK,KC_DELETE,KC_DOWN,KC_4,KC_5,KC_6,KC_KP_MINUS,KC_KP_ENTER,KC_TRANSPARENT,KC_AMPR,KC_1,KC_2,KC_3,KC_KP_SLASH,SFT_T(KC_KP_PLUS),KC_KP_0,KC_KP_0,KC_KP_DOT,KC_EQUAL,KC_KP_ENTER,RGB_TOG,RGB_SLD,RGB_MOD,RGB_HUD,KC_TRANSPARENT,KC_TRANSPARENT),
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  ^~~~~~~
keyboards/ergodox_ez/ergodox_ez.h:145:33: note: in definition of macro 'LAYOUT_ergodox'
     { k08, k18, k28, k38,KC_NO, k58 },   \
                                 ^~~
keyboards/ergodox_ez/keymaps/damir/keymap.c: In function 'process_record_user':
keyboards/ergodox_ez/keymaps/damir/keymap.c:73:10: error: 'EPRM' undeclared (first use in this function); did you mean 'EPEN'?
     case EPRM:
          ^~~~
          EPEN
keyboards/ergodox_ez/keymaps/damir/keymap.c:73:10: note: each undeclared identifier is reported only once for each function it appears in
keyboards/ergodox_ez/keymaps/damir/keymap.c:79:10: error: 'VRSN' undeclared (first use in this function)
     case VRSN:
          ^~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_ergodox_ez_damir/keyboards/ergodox_ez/keymaps/damir/keymap.o] Error 1
make: *** [ergodox_ez:damir] Error 1
Make finished with errors
```
This time, I'm stuck...
Anyone can help?

Thank you in advance,

Damir.

drashna, 2018-07-08:
If you run `git revert 9c2dde98e2b963704eb7cb87f6c53c52599fba53 -m 1`, it will revert the change that broke the RGB Underglow (shine) functionality. 

Also, could you post your keymap.c file?  Link to it specifically (eg, if you have a github repo of it).  That way, we can take a look at what is going on.

zzz2496, 2018-07-08:
@drashna How do I make a git repo? As for the Shine LED, I'll leave it be for now :) It's not that important (yet).
Here's my current keymap.c file
[keymap.c](https://github.com/qmk/qmk_firmware/files/2173896/keymap.c.txt)

Thanks in advance,

Damir.

drashna, 2018-07-08:
@zzz2496 https://docs.qmk.fm/#/getting_started_github

Ideally, you'd want to fork QMK first, then clone from that repo.  Once you've done that, you can push to your repo, and even submit "Pull Requests" (adding your keymaps/code to the official repo).

drashna, 2018-07-08:
Your rules.mk should look like this: 
```

BOOTMAGIC_ENABLE = no  # Virtual DIP switch configuration(+1000)
MOUSEKEY_ENABLE  = yes # Mouse keys(+4700)
EXTRAKEY_ENABLE  = yes # Audio control and System control(+450)
CONSOLE_ENABLE   = no  # Console for debug(+400)
COMMAND_ENABLE   = yes # Commands for debug and configuration
CUSTOM_MATRIX    = yes # Custom matrix file for the ErgoDox EZ
NKRO_ENABLE      = yes # USB Nkey Rollover - if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ#nkro-doesnt-work
UNICODE_ENABLE   = yes # Unicode
SWAP_HANDS_ENABLE= yes # Allow swapping hands of keyboard
SLEEP_LED_ENABLE = no
API_SYSEX_ENABLE = no

RGBLIGHT_ENABLE = no
RGB_MATRIX_ENABLE = no
TAP_DANCE_ENABLE = yes
```
If you have/are able to revert the commit as above, then you can set `RGBLIGTH_ENABLE = yes`

In your keymap.c, you have this:
```c
//In Layer declaration, add tap dance item in place of a key code
TD(TD_ESC_CAPS)
```
This needs to be removed.  Once you've removed that, you'll be able to compile the firmware.
This block is meant to be more of an example, and not placed literally. But I can see how this could be confused.  

That said, I've compiled the firmware on my system, so you should be able to have RGB working.

https://drive.google.com/open?id=1xSAW9ZnRgbY4johYNSct_64gFbgyaZpA

zzz2496, 2018-07-08:
@drashna, QMK cloned - it's at https://github.com/zzz2496/qmk_firmware... 

rules.mk created inside the "damir" directory. Successfully revert the repo. Now the Tap Dance is working - WITH RGB, YAY.

closed 2018-07-08 by zzz2496

drashna, 2018-07-08:
Awesome, glad to hear it!  And enjoy!