qmk/qmk_firmware #3366: LCTL_T with KC_CAPS doesn't work as intended

squarefrog, 2018-07-10:
I'm trying to set the a key to function as a `LCTL` when held, but `CAPS` when tapped. To do this I use `CTL_T(KC_CAPS)`, however it doesn't function as I'd expect. It seems to toggle `CAPS` on and off immediately.

If I use a different key for `CAPS` I can see that tapping the mapped control key turns it off.

Is this a limitation of this `CTL_T`, or is there another way I can do this?

zzz2496, 2018-07-10:
Hi,
What keyboard are you using?
I tried the same thing, but I use left shift instead.
I was successful with several quirks.

Here's what I did:

1. You need to enable Tap Dance - https://docs.qmk.fm/#/feature_tap_dance
2. Add this into your rules.mk file
`TAP_DANCE_ENABLE = yes`
3. Add this under `#include "keymap_nordic.h"` in your keymap.c file
```
//TAP DANCE
int IS_CAPS = 0;
enum {
  TD_SHIFT_CAPS = 0,
};

void dance_fc_finished (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
    register_code (KC_LSHIFT);
  }else if (state->count == 2) {
    if (IS_CAPS == 0) {
      register_code (KC_LSHIFT);
      IS_CAPS = 1;
      ergodox_right_led_3_on();
    }else{
      unregister_code (KC_LSHIFT);
      IS_CAPS = 0;
    }
  }
}

void dance_fc_reset (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
    unregister_code (KC_LSHIFT);
    ergodox_right_led_3_off();
  }
}

//Tap Dance Definitions/
qk_tap_dance_action_t tap_dance_actions[] = {
  [TD_SHIFT_CAPS]  = ACTION_TAP_DANCE_FN_ADVANCED(NULL, dance_fc_finished, dance_fc_reset),
// Other declarations would go here, separated by commas, if you have them
};
//In Layer declaration, add tap dance item in place of a key code
//TAP DANCE
```
4. In your keymap, put TD(TD_SHIFT_CAPS) on the left shift.

Single tap on left shift will send normal shift, double tap left shift will hold the button until you press it again. I never tried with ctrl. You can try and modify my code, see if it will fit your needs.

Damir.

squarefrog, 2018-07-10:
Thanks for the reply. That wasn’t quite what I wanted, I wanted to have control work as control when held, but if tapped, it should toggle caps lock. The reason is I remap caps lock to control, but sometimes it’s be nice to still have caps lock without losing a button. 

However in the mean time, I think I like the idea of a dual action shift, so that’s definitely tide me over until I can play around more. Thanks again.

zzz2496, 2018-07-11:
@squarefrog You're most welcome. You can easily modify my source code to fit your needs. There is a latching capslock support in QMK IIRC, so you can use it along with Tap Dance on a CTRL assigned button.

closed 2018-10-22 by drashna