qmk/qmk_firmware #3370: Question about MOUSEKEY_WHEEL_DELAY

zzz2496, 2018-07-11:
Hi everyone,

@drashna 
@ezuk 
@jackhumbert 

I've been playing with QMK for a week now, and starting to explore it's features. This time I think I  found a bug, it relates to MOUSEKEY_WHEEL_DELAY. In my config, I put 165535 in it, and the mouse wheel button does not change in behavior, and nothing I can do to make the scroll wheel slow when I set MOUSEKEY_INTERVAL to around 10 or 20. I like low MOUSEKEY_INTERVAL, it gives me smooth mouse movement, but I hate it as it will scroll the content in my windows VERY VERY VERY FAST. And I haven't been able to achieve smooth mousing with QMK. Here are my mousekey parameters in config.h file:

```
// Sets good default for the speed of the mouse.
#undef MOUSEKEY_INTERVAL
#undef MOUSEKEY_DELAY
#undef MOUSEKEY_TIME_TO_MAX
#undef MOUSEKEY_MAX_SPEED

#define MOUSEKEY_INTERVAL       25
#define MOUSEKEY_DELAY          0
#define MOUSEKEY_TIME_TO_MAX    30
#define MOUSEKEY_MAX_SPEED      6

#undef MOUSEKEY_WHEEL_MAX_SPEED
#undef MOUSEKEY_WHEEL_TIME_TO_MAX
#undef MOUSEKEY_WHEEL_DELAY

#define MOUSEKEY_WHEEL_MAX_SPEED    0
#define MOUSEKEY_WHEEL_TIME_TO_MAX  255
#define MOUSEKEY_WHEEL_DELTA        0
```
I even added MOUSEKEY_WHEEL_DELTA, and play around using that variable, it speeds up the scrolling EVEN MORE when set to high value. The lowest value of 0 doesn't do anything. Anyone knows why is it doing that?

Thanks,

Damir.

drashna, 2018-07-20:
I'm not familiar with those code, so I'm not sure how much help I'd be. 

MOUSEKEY_WHEEL_DELAY doesn't appear to actually be used, so ... not sure about that.

https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/mousekey.c

zzz2496, 2018-07-20:
@drashna finally you're back !!!
Yeah, I searched into the code, and I can't find it to be used anywhere. Maybe it's someone's variable on another branch of code that didn't get committed into the master branch?

I'm still trying to understand the C (or is it C++?) code to do things that I wanted to do.
Almost a month in, and I still can't have a fixed layout, oh well...

Maybe someone knows about this? How to slow down the scroll wheel independent of the mouse movement.

By the way, I finally got the CONSOLE, COMMAND, TAP DANCE, and AUTO SHIFT enabled - and now I'm ruined (in a good way). I cannot use another simple/dumb/non-programmable (non QMK at least for now) keyboard... This is a bit frustrating...

PR #6685 (2020-03-13): Decouple mouse cursor and mouse wheel in accelerated mode
## Description

In accelerated mode, MOUSEKEY_INTERVAL is used for both the cursor and the wheel. Since the step for the wheel is way higher than the step for the cursor, this results in the wheel being almost unusable. See issue #3370 for example.

In this change, I decoupled the cursor and the wheel behaviors (using the decoupling pattern in the constant mode). I also introduced 2 settings:

- MOUSEKEY_WHEEL_DELAY
- MOUSEKEY_WHEEL_INTERVAL

which are the equivalent of MOUSEKEY_DELAY and MOUSEKEY_INTERVAL. Users can therefore choose di [... cut]
closed 2020-03-13 by jackhumbert