qmk/qmk_firmware #3387: Hand wired Dactyl w/ Pro Micros compiles and flashes but shows up as unknown device

Krakob, 2018-07-12:
Hello! Currently building a Dactyl. As the title says, it's using Pro Micros and the firmware compiles and flashes correctly. Plugging it in on Windows 7 shows it as a bunch of devices, with one subdevice as unknown and reporting error code 43. On Linux it doesn't show up with lsusb at all. Neither take any input from it. I'm using serial for the cross communication, but I don't think that's an issue here.

My firmware code is at https://gitlab.com/Krakob/Dactyl-qmk/tree/master/keyboards/Dactyl

It's the version Crystalhand recommended when he ran the group-buy back in February. I've tried popping it into the latest version of qmk as well, but I'm getting the same results there.

Thanks in advance!

Krakob, 2018-07-14:
For reference, how the device shows up in Windows:
[image]

kdb424, 2018-07-14:
Are you flashing on Linux or Windows? If on Linux, do you have ModemManager installed? This can easily corrupt the flash.

Krakob, 2018-07-14:
Couldn't save the micros, but luckily I had exactly four when beginning this project. Thanks so much! :)

closed 2018-07-14 by Krakob