qmk/qmk_firmware #3391: ps2avrgb

qiaokang0, 2018-07-13:


drashna, 2018-07-13:
@mechmerlin Suggestions?

mechmerlin, 2018-07-13:
Can you post the layout you made on kbfirmware? If you did not make it correctly, this is exactly what would happen.

qiaokang0, 2018-07-14:
["Esc","!\n1","@\n2","#\n3","$\n4","%\n5","^\n6","&\n7","*\n8","(\n9",")\n0","_\n-","+\n=","~\n`","Backspace"],
[{w:1.5},"Tab","Q","W","E","R","T","Y","U","I","O","P","{\n[","}\n]",{w:1.5},"|\n\\"],
[{w:1.75},"Caps Lock","A","S","D","F","G","H","J","K","L",":\n;","\"\n'",{w:2.25},"Enter"],
[{w:2},"Shift","Z","X","C","V","B","N","M","<\n,",">\n.","?\n/","Shift","up","del"],
[{w:1.75},"Ctrl",{w:1.25},"Win",{w:1.25},"Alt",{w:1.25},"Fn",{w:3},"space",{w:1.25},"Fn",{w:1.25},"Alt","App","left","down","right"]

i check my controller is atmega32u4 , i don't know if it is supported in the same way like others. and i use the default wiring after i paste layout above , maybe the wiring cause the problem? should i check the bootmapper about the rows and cols?

mechmerlin, 2018-07-18:
The bface is not supported by QMK at this time. Your best bet would be to continue using bootmapperclient until support has been added.

closed 2018-10-22 by drashna