qmk/qmk_firmware #3401: bfake firmware not working correctly with my bfake.

sm0ggles, 2018-07-14:
So I'm not sure if this is a problem on my end or not, but when I try to flash the default bfake build the keys are all messed up. Idk if this is a problem with the matrix or what but I've been trying to figure it out for hours. If I flash the ps2avrGB default firmware everything works correctly (except my split backspace setup) Not sure what's going on but any help would be wonderful. 

Thanks,
sm0g

noroadsleft, 2018-07-14:
More details? There's not a lot to go on here.

Are the keys all one row or column off? Are they scrambled around (one key moves next door, another key moves across town)?

drashna, 2018-07-14:
what command are you using specifically to make the board? 

And it sounds like the matrix is very different.  

And maybe this is the facew?

sm0ggles, 2018-07-14:
@drashna I'm just using make bfake:default to make the default hex for the bfake. I'm thinking the matrix is completely different as it appears to be the same as the bmini since that hex file works fine on my bfake. I'm just thinking that the PCB I'm using is different than the bfake that was used to build the qmk files. I hope that makes sense. I used to have a faceW but I don't have that anymore, I wish I did so I could test and see if the matrix is the same between the two PCBs. This is the PCB I'm currently trying to flash. https://bit.ly/2JnM1yT the bface northern variant. 

I think the problem is with the matrix and I'm not sure how to go about making the correct matrix for my PCB.

drashna, 2018-07-14:
so `make bmini:default` works on the board?

sm0ggles, 2018-07-14:
I tried make:ps2arvGB:default and that worked. But the problem is I'm not entirely sure how I'd add my split backspace to that layout. I'm sorry, I'm still learning how QMK works.

drashna, 2018-07-14:
.... That magic is here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/ps2avrGB/bmini.h#L23-L39

But ... this is not simple stuff

sm0ggles, 2018-07-14:
Yeah, I was looking at that last night and was trying to make sense of it. Looking at the QMK docs to try and figure it out but I gave up because I figured I'd need to use a multimeter or something to map out the matrix. I know it's at least close to the bmini matrix. I was really hoping that the bfake build would work with my pcb. lol.

noroadsleft, 2018-07-15:
I'm still looking this over and I'm still puzzled, but not giving up.

- `make ps2avrGB:default` works
  - Split Backspace doesn't work, which I can already see the reason why - the matrix there is configured for a 2u Backspace key
  - ps2avrGB matrix is identical to that of bmini
- `make bfake:default` does not work; most keys are out of place.
  - Makes sense to me, as the matrix appears to be completely different.

I'm still looking.

noroadsleft, 2018-07-15:
I've looked at a few pages and asked someone knowledgeable, and I'm thinking the PCB may be a B.fa**c**e (or a really good copy) rather than a B.fa**k**e.

I've been comparing the files for `ps2avrGB`, `bmini`, `bfake` and `facew` (more on that in a minute).

The B.fake's matrix is identical to that of the FaceW. The matrix in ps2avrGB is a B.mini matrix.

The matrices for ps2avrGB/B.mini are completely different to that of B.fake/FaceW.

I also compared the photos from @sm0ggles' AliExpress listing to photos on the [shop page for the Winkeyless.kr B.face R](https://winkeyless.kr/product/b-face-r-x2-pcb/). The photos on AliExpress seem to my eyes to be the Winkeyless PCB with components pre-soldered (the Winkeyless photos are of bare PCBs). If @sm0ggles's PCB is actually a B.face R, it wouldn't surprise me at all if the matrix was similar to that of the B.mini, which according to him, it is.

sm0ggles, 2018-07-15:
@noroadsleft yeah, I'm starting to think that too, which is surprising since it only cost me like $35 USD. All I want is a darn split backspace and I have no idea how to do it. Lol.

closed 2018-10-22 by drashna