qmk/qmk_firmware #3407: Adding Adafruit Bluefruit LE SPI Friend

leonpg2003, 2018-07-15:
I bought Adafruit Bluefruit LE SPI Friend to add Bluetooth function to my keyboard but I cant get it to work. When the keyboard is plugged via USB the keypress gets inputted into my computer but when i try to connect via Bluetooth it doesn't work. I am using a teensy 2.0. 
https://gist.github.com/leonpg2003/bb883d7ae929055e82050bbc45231a87

myoung34, 2019-12-04:
I'm hitting this also with the Adafruit Feather 32u4.
It works over USB, but nothing over BT.

Im using a fork of the meira firmware on a custom handwire.

`rules.mk`:

```
MCU = atmega32u4
F_CPU = 8000000
BLUETOOTH_ENABLE = yes
BLUETOOTH = AdafruitBLE
```

myoung34, 2019-12-04:
Nevermind, its working now after a forced reset.
The rules.mk above are the key to getting the feather 32u4 working with SPI

jackhumbert, 2020-03-17:
@leonpg2003 did a force reset work for you in this case with the rules.mk above?

fauxpark, 2020-03-17:
This issue is probably fixed as of https://github.com/qmk/qmk_firmware/pull/6671

jackhumbert, 2020-03-18:
Ah, nice - closing for now then!

closed 2020-03-18 by jackhumbert