qmk/qmk_firmware #3448: Tap dance not working on  Let's Split - Vitamins Included

kdb424, 2018-07-20:
I have a working levinson which is the same layout as a let's split working perfectly. However, when trying to use my layout from the levinson on the let's split, it's ignoring all of my tap dance keys. I have left the other 2 files (the rules and config) at bone stock on the let's split minus of course turning on tap dance in the rules, and setting a tapping term in the config, which once again, works on my levinson perfectly. Here is my keymap for reference.

https://github.com/kdb424/qmk_firmware/blob/master/keyboards/levinson/keymaps/kdb424/keymap.c

As I try more things, my files for the board that I'm working on will be here, though at the time of posting this, I have yet to push my current changes.

https://github.com/kdb424/qmk_firmware/tree/master/keyboards/vitamins_included/keymaps/kdb424

donpark, 2018-07-20:
Possibly related: `MT` and `TT` macros have no effect on NovelKeys version of Vitamins Included.

kdb424, 2018-07-20:
I haven't been able to get MO to work either. I'm pretty much stuck on default layout right now.

drashna, 2018-07-20:
This is why: https://github.com/qmk/qmk_firmware/blob/master/keyboards/vitamins_included/rev1/config.h#L86

Not sure why this is included, in the first place....

But if you add `#undef NO_ACTION_TAPPING` to your `config.h`, it should fix the issue.

kdb424, 2018-07-20:
Have I ever told you that you are my god? Everything solved.

closed 2018-07-20 by kdb424

donpark, 2018-07-20:
Unfortunately, I got the `Your file is too big! 29534/28672` error after adding `#undef NO_ACTION_TAPPING` to `config.h`. But adding the following to `rules.mk` reduced hex file size down to `18530`.

```
AUDIO_ENABLE = no
RGBLIGHT_ENABLE = no
```

And, finally I can tap dance again! Thanks @drashna

drashna, 2018-07-20:
Aaah, that must be why that was disabled.  

I have a better option, actually.

In your rules.mk, re-enable those options (trust me). 
Then add this to your rules.mk: 
```makefile
EXTRAFLAGS        += -flto
```

This is enableds "link time optimization" and can significantly reduce the compile size (at the expense of a couple of features and compile time)

Then, in your config.h file, add:
```h
#if !defined(NO_DEBUG) && !defined(CONSOLE_ENABLE)
#define NO_DEBUG
#endif // !NO_DEBUG
#if !defined(NO_PRINT) && !defined(CONSOLE_ENABLE)
#define NO_PRINT
#endif // !NO_PRINT
#define NO_ACTION_MACRO
#define NO_ACTION_FUNCTION

#define DISABLE_LEADER
```

This disables debugging and HID printing, if the console is not enabled (it's disabled by default).  

Additionally, this disables the `action_get_macros` and `fn_actions` functions (eg, M(x) and KC_FNx).  But these are deprecated, anyways.  So there should be no real loss.

Duckle29, 2018-07-27:
Thanks for this! Both the feedback and the fix :)

Feel free to tag me on any vitamins_included related issues, as I'm the one working on that :) I'll get these changes into the stock code for the vitamins_included :)