qmk/qmk_firmware #3454: Add support for other ISSI LED drivers

VinnyCordeiro, 2018-07-20:
On [ISSI FxLED Driver page](http://www.issi.com/US/product-analog-fxled-driver.shtml) there is now sample code for the evaluation boards of a lot of their drivers (STM32 microcontrollers). Some have even Arduino code for them, which could be ported much more quickly. That would help adding support to larger drivers, like the IS31FL3733 (up to 64 RGB LEDs) and the new IS31FL3741 (117 RGB LEDs!).

Wilba6582, 2018-07-20:
I'll be writing a driver for IS31FL3736 soon, as it's what I'm using for a few PCBs I'm working on now (with mono LEDs). It's very similar to the IS31FL3733. One of the cool features is detecting missing/shorted LEDs, so the firmware can automatically configure itself to the LEDs that have been installed.

drashna, 2018-10-22:
I think we added generalization to the rgbmatrix driver stuff, so this should be handled better now.

closed 2018-10-22 by drashna