qmk/qmk_firmware #3456: error: excess elements in array initializer

Merlin04, 2018-07-20:
https://github.com/Merlin04/qmk_firmware/tree/master/keyboards/fractal
I based this keymap off of the Preonic keymap. When I try to build it, I get these errors:
```make
qmk_firmware> make fractal:default
QMK Firmware 0.6.75
Making fractal with keymap default

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/fractal/fractal.c                                                              [OK]
Compiling: keyboards/fractal/keymaps/default/keymap.c                                              In file included from keyboards/fractal/keymaps/default/keymap.c:19:0:
keyboards/fractal/fractal.h:17:3: error: excess elements in array initializer [-Werror]
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:54:13: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_QWERTY] = LAYOUT_preonic_grid( \
             ^
keyboards/fractal/fractal.h:17:3: note: (near initialization for ‘keymaps[0]’)
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:54:13: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_QWERTY] = LAYOUT_preonic_grid( \
             ^
keyboards/fractal/fractal.h:17:3: error: excess elements in array initializer [-Werror]
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:75:14: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_COLEMAK] = LAYOUT_preonic_grid( \
              ^
keyboards/fractal/fractal.h:17:3: note: (near initialization for ‘keymaps[1]’)
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:75:14: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_COLEMAK] = LAYOUT_preonic_grid( \
              ^
keyboards/fractal/fractal.h:17:3: error: excess elements in array initializer [-Werror]
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:96:13: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_DVORAK] = LAYOUT_preonic_grid( \
             ^
keyboards/fractal/fractal.h:17:3: note: (near initialization for ‘keymaps[2]’)
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:96:13: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_DVORAK] = LAYOUT_preonic_grid( \
             ^
keyboards/fractal/fractal.h:17:3: error: excess elements in array initializer [-Werror]
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:117:12: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_LOWER] = LAYOUT_preonic_grid( \
            ^
keyboards/fractal/fractal.h:17:3: note: (near initialization for ‘keymaps[3]’)
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:117:12: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_LOWER] = LAYOUT_preonic_grid( \
            ^
keyboards/fractal/fractal.h:17:3: error: excess elements in array initializer [-Werror]
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:138:12: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_RAISE] = LAYOUT_preonic_grid( \
            ^
keyboards/fractal/fractal.h:17:3: note: (near initialization for ‘keymaps[4]’)
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:138:12: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_RAISE] = LAYOUT_preonic_grid( \
            ^
keyboards/fractal/fractal.h:17:3: error: excess elements in array initializer [-Werror]
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:159:13: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_ADJUST] = LAYOUT_preonic_grid( \
             ^
keyboards/fractal/fractal.h:17:3: note: (near initialization for ‘keymaps[5]’)
   { K400,  K401,  K402,  K403,  K404,  K405,  K406,  K407,  K408,  K409,  K410,  K411 }  \
   ^
keyboards/fractal/keymaps/default/keymap.c:159:13: note: in expansion of macro ‘LAYOUT_preonic_grid’
 [_ADJUST] = LAYOUT_preonic_grid( \
             ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:357: recipe for target '.build/obj_fractal_default/keyboards/fractal/keymaps/default/keymap.o' failed
make[1]: *** [.build/obj_fractal_default/keyboards/fractal/keymaps/default/keymap.o] Error 1
Makefile:529: recipe for target 'fractal:default' failed
make: *** [fractal:default] Error 1
Make finished with errors
```
I have no idea what I did wrong, I've looked through all the files and I can't find any mistakes.

drashna, 2018-07-20:
https://github.com/Merlin04/qmk_firmware/blob/master/keyboards/fractal/config.h#L15

You've defined 4 rows, but you've used 5.   Change that to 5, and it will compile. 

(also, awesome to see you working on this here!)

noroadsleft, 2018-07-20:
Additional things I saw, but unrelated to the issue @drashna solved above:

Firstly, https://github.com/Merlin04/qmk_firmware/blob/3895b54e96ad88bc23d7c24e43ef7cfbab200a02/keyboards/fractal/config.h#L20

You have `#define MATRIX_COLS 12` on Line 16, but there's only 11 pins set on Line 20.

Secondly, https://github.com/Merlin04/qmk_firmware/blob/3895b54e96ad88bc23d7c24e43ef7cfbab200a02/keyboards/fractal/fractal.h#L31

You need to insert a `KC_NO` element between `K404` and `K406` (assuming your 2u spacebar is on the 7th column of the matrix), Without it, your last row is one column short, which will lead to misplaced keys.

Merlin04, 2018-07-21:
Thanks, that worked.

closed 2018-07-21 by Merlin04