qmk/qmk_firmware #3467: JJ40 underglow LEDs 1/5 working

danicedenv, 2018-07-22:
I got myself a jj40 for my first build. I plugged in the pcb, tested every switch contact and it worked fine. The only issue was that only 1 underglow led was working. I decided to use qmk configurator to make my own layout. I flashed it onto the board and everything worked great except I still had the same issue. I decided to leave it and try later and when I plugged it in the leds were working fine, the only issue were the keys I use to setup the rgb they stopped working. After plugging it back in one more time it went back to only one led working. I am pretty new to this so I have 0 clue what the problem is. I could really use some help! Thanks in advance

noroadsleft, 2018-07-22:
I know nothing personally about this issue, but #3199 is likely related.

drashna, 2019-11-21:
I suspect that this may be working now, since all of the ps2avrGB boards have been overhauled.

closed 2019-11-21 by drashna