qmk/qmk_firmware #3471: Random caps lock enable

333fred, 2018-07-23:
Ever since I rebuilt and redeployed my firmware on Tuesday 7/17, I've been getting random enables of CAPS lock that are, as far as I can tell, not being triggered intentionally on my part (especially as my keyboard does not have caps lock mapped). I haven't been able to find a consistent repro for this issue, but hopefully on Monday I can start using git bisect and just using my keyboard to attempt to narrow down the commit that started causing this. FYI @drashna @r2d2rogers, who've also reported seeing this behavior recently.

333fred, 2018-07-26:
Alright, after a few days of testing, I think that I've determined that the regression occurred sometime between 6fbd0c5 and 5477bf3. I've been on 6fbd0c5 for the past few days with no issues (it's hard to prove a negative :) ), so I'll start a bisect between those two commits and see if I can determine the actual cause.

333fred, 2018-08-15:
I haven't seen it after updating to the latest qmk master, but I was seeing it pretty regularly before I updated. I'm wondering if it was fixed by another commit at this point.

drashna, 2018-10-22:
Hasn't happened since.  Safe to close, I think.

closed 2018-10-22 by drashna