qmk/qmk_firmware #3485: Need documentation about international layouts

skullydazed, 2018-07-25:
We have a lot of users who come into discord to ask about how to work with international layouts. We should write a page documenting how that works.

* Short description how letters appear when you type. switch -> scancode -> computer -> language mapping -> letter on screen
* Explain that due to historical reasons US ANSI is the "default"
* Explain how to use language specific includes in `quantum/keymap_extras`

ThomDietrich, 2018-07-25:
Hello,
as a fresh Ergodox owner located in Germany I have similar questions and would welcome such a page. Is a *raw* pile of information present somewhere?

skullydazed, 2018-07-25:
I think the closest we have are the includes here: https://github.com/qmk/qmk_firmware/tree/master/quantum/keymap_extras

If you're working on this, or just trying to figure out how to make it happen with your own keyboard, right now the best place to get more info is [discord](https://discord.gg/Uq7gcHh), people there have varying levels of knowledge.

ThomDietrich, 2018-07-25:
* https://github.com/qmk/qmk_firmware/blob/master/docs/how_keyboards_work.md

* https://github.com/qmk/qmk_firmware/issues/520
* https://www.reddit.com/r/olkb/comments/4vrie0/is_there_a_way_to_make_the_plancks_layout/

ubone, 2019-08-03:
is having a cyrillic layer even possible?

MetalDent, 2019-10-22:
Hi, anyone working on this issue?

sakshamtaneja21, 2019-10-26:
there are so many typos in this file

jackhumbert, 2020-03-17:
@fauxpark is working on some docs that will help with this.

fauxpark, 2020-07-06:
This is done as of #9194:
https://docs.qmk.fm/#/reference_keymap_extras

closed 2020-07-06 by fauxpark