qmk/qmk_firmware #3533: Raise? Lower? Adjust?

pvinis, 2018-07-31:
hello. I've been using qmk for around two years and personally I had a qwerty layer, a system control and a third that I don't remember while writing now. now I see many keymaps using a base like qwerty and then three more, raise, lower and adjust.
is that the "best practices" now? where does it come from? what are the ideas behind it?
this issue is more for discussion I guess.
thanks.

drashna, 2018-07-31:
Well, there is no "standard".  You can use whatever you want. No limits. :)

That said, I believe that Jack developed the tri layer stuff (raise/lower/adjust), and those would mean that they originate from the OLKB boards (planck, preonic, etc).  

Personally, I use them because it allows for you to use two layer buttons to access 3 layers, not just two.  So it's more efficient for boards that have fewer keys.   Additionally, since the addition of `layer_state_set_user` and the `update_tri_layer_state` functions, you can use any layer keycode to activate it, rather than just custom codes.  So you could use `LT` to trigger the tri layer stuff. 

But as I said, this isn't necessary, at all.  Many boards have a "function" layer, that has some of the custom codes.  But even this isn't necessary.  

My best suggestion is to look around and see what people are using, and use whatever fits your needs best.

wanleg, 2018-07-31:
Planck's _ADJUST layer is fine for "one-touch" keys (e.g. on/off, layer select).
but on a [pretty small board](https://github.com/qmk/qmk_firmware/tree/master/keyboards/gherkin/keymaps/wanleg), it's kinda painful to have to hit TWO keys simultaneously just to access another layer; especially if you're going to have to chord on top of that (e.g CTRL-ALT-DEL).
As a "quick cheat" on a larger board, it has its uses...

drashna, 2018-07-31:
@wanleg on a 30%, yeah, I can definitely see that being a problem! 

@pvinis Yup, if the raise and lower layers are active, then it shifts to the Adjust layer.   But only if you have it configured to do so (eg, this requires the `update_tri_layer` or `update_tri_layer_state` functions to be called at some point).   I use it on all of my boards, as it makes things a lot simpler, and a lot more uniform. 

>another idea I had in my old keymaps, is to have a base keymap with the layer keys plus the basic like esc ctl caps gui tab etc, so all the surrounding and then on top I would have a qwerty that has only the letters and everything else transparent. this way I could play with a new layout and keep my surrounding keymap.

Yeah, that's entirely doable.  Or you could have a layer that changes these too. 

I actually have a layer that changes my One Shot Mods to regular mods, since they can cause issues in some cases. 

But if it works for you, then by all means!

closed 2018-10-22 by drashna

bigwheel, 2020-09-27:
I sought the origin of raise / lower / adjust layer in qmk_firmware and I think I can find it.

* Raise and Lower layer first commit: https://github.com/qmk/qmk_firmware/commit/4a8b9b8a2e2ef98827109be08289e79fb0695d54
* Adjust layer first commit: https://github.com/qmk/qmk_firmware/commit/e6a41e43aa524ae78aecb2eeeeca55c850d05cdd

Raise and Lower layer commit has done by jack but Adjust layer is CommandLineDesign at least in qmk_firmware repo.