qmk/qmk_firmware #3551: Niu mini planck build is giving Your file is too big error

crisalm29, 2018-08-02:
Im trying to build the niu_mini:planck but I got an error with the niu_mini_planck.hex when I run the 

make niu_mini:planck

I have an working example of the niu_mini_planck.hex file but that I just got from Internet, I cannot build my own,

I have the repo up-to-date and I am using msys to build

Falyure, 2018-08-02:
I have the same issue. I'm also on msys. I've tried to delete some keylayours I didn't need, but it didn't changed anything.

Falyure, 2018-08-02:
you can solve it by adding:
```
#if !defined(NO_DEBUG) && !defined(CONSOLE_ENABLE)

#define NO_DEBUG

#endif // !NO_DEBUG

#if !defined(NO_PRINT) && !defined(CONSOLE_ENABLE)

#define NO_PRINT
```
#endif // !NO_PRINT

#define NO_ACTION_MACRO

#define NO_ACTION_FUNCTION

#define DISABLE_LEADER
#define MIDI_BASIC

into the config.h header.

crisalm29, 2018-08-02:
Thanks that solve the issue!

closed 2018-08-02 by crisalm29