qmk/qmk_firmware #3555: Backlight keycodes not working on XD75

ArpinFidel, 2018-08-02:
Hi, I'm new to qmk. I'm trying to add backlight functionality to my XD75 but the default keycodes (BL_TOGG, BL_INC, etc) doesn't work. Is this because XD75 uses the f5 pin?

I got around the issue by using the keycaps_led_on and off from xd75.c but these don't allow me to change the backlight brightness. Is there some other way to have brightness control? 

I wanted to see the code for the default keycodes to find the problem but I couldn't even find the file where it's defined and the documentation wasn't much help.

wanleg, 2018-08-07:
why are you using BL_? i use [RGB_ instead](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_rgblight.md) and don't have any issues
here's my [layout](https://github.com/wanleg/qmk_firmware/tree/WanlegPersonal/keyboards/xd75/keymaps/wanleg) if you want to compare

wanleg, 2018-08-07:
have you seen this [reddit post](https://www.reddit.com/r/MechanicalKeyboards/comments/7tjql1/xd75_backlighting_flickering/) already?

ArpinFidel, 2018-08-09:
ooh thanks. i haven't actually. only searched on r/olkb lol. that one is about the same thing as issue #2226 though. it was fixed a while back.

i found another issue similar to this even on a supported pin #3566 and the flickering is apparently documented already so this should probably be closed.

closed 2018-08-09 by ArpinFidel