qmk/qmk_firmware #3565: Updates on Linux USB in docs/hardware_avr.md

peremen, 2018-08-04:
On the section "config.h -> Hardware Configuration" this phrase does not correctly reflect what is in Linux:

> ?> On Linux these values will not be visible in lsusb, since Linux takes that information from the list published by the USB-IF.

First of all, lsusb's list is not coming directly from USB-IF, but maintained by USB ID Repository (http://www.linux-usb.org/usb-ids.html). Also that value is visible in either executing `dmesg` after plugging in the keyboard:
```
[ 7315.639276] usb 1-2.4.1: new full-speed USB device number 9 using xhci_hcd
[ 7315.753726] usb 1-2.4.1: New USB device found, idVendor=feed, idProduct=6060
[ 7315.753732] usb 1-2.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7315.753736] usb 1-2.4.1: Product: GH60
[ 7315.753740] usb 1-2.4.1: Manufacturer: geekhack
```

or running `lsusb` with a switch `-v`. Following example is taken from my GH60:

```
Bus 001 Device 009: ID feed:6060  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0xfeed 
  idProduct          0x6060 
  bcdDevice            0.01
  iManufacturer           1 geekhack
  iProduct                2 GH60
```

So I suggest rewriting that sentence as follows:

> ?> On Linux these values will not be visible in lsusb by default, since Linux takes the information from the list maintained by USB ID Repository is presented by default. lsusb will show the information reported by the device when executed with -v option. It is also present in kernel logs after plugging in the device.

kdb424, 2018-08-06:
I agree with this. I have sorted this out for my boards, but friends that I have that also use Linux have asked me about similar "issues". Clearing up the docs makes users lives easier, and frees us up to help with more serious issues.

drashna, 2018-10-22:
It looks like this note was removed altogether, so closing this issue.

PR #4202 (2018-10-22): Update hardware config warnings to be more accurate
Closes #3565
commit ab3fbfdb (2018-10-22): Update hardware config warnings to be more accurate

Closes #3565
closed 2018-10-22 by jackhumbert