qmk/qmk_firmware #3572: Single Use Modifiers

dankMATCHA, 2018-08-05:
Hey guys, 
I'm looking for a way to use my shift such that it only modifies the first key after it fires. Similar to One Shot but I need it to work even if the shift key is still pressed.

Background: I find myself to be a little laggy/lazy when typing quickly and lots of "The"s and similar words end up as "THe" or even "THE". 

The concept behind OneShot makes sense but it doesn't help me with the laggy shift key. I was thinking about writing a little timer scenario that only lets shift modify for a small window but I was hoping there was a more elegant solution I missed somewhere.

Thanks for the help!

MxBlu, 2018-08-06:
Something like this in the process_record_user (or kb):

```c
shift_down = 0; // record last checked status of shift
// 0 - shift not down
// 1 - shift down
// 2 - shift down, but another key already pressed

process_record_user(keycode, record) {
   if (keycode == LSHFT/RSHFT) { // key is shift
      if (record->event.pressed) { // set shift status
         shift_down = 1;
      } else {
         shift_down = 0;
      }
   } else if (shift_down == 1) { // key is not shift, and shift was pressed 1 keys ago
       shift_down = 2; // unset shift next time
   } else if (shift_down == 2) { // key is not shift, shift was pressed 2 keys ago
       shift_down = 0; // reset shift to unpressed
       unregister_code(LSHFT);
       unregister_code(RSHFT);
   }
}
```
I wrote it in pseudo code, I know its a touch confusing but I hope you can understand it. Basically, shift_down is used as a 3 state variable being either "unpressed/reset" where the shift is not pressed or considered unset, "pressed 1 key ago" where it was the last key pressed, and the current key should be shifted, and "pressed 2 keys ago" where it was pressed 2 keys ago, and the current key should be unshifted, and the shift status reset to unpressed.

dankMATCHA, 2018-08-07:
It's taking me a while to work through the actual implementation of the code but the idea is exactly what I was looking for!

closed 2018-08-07 by dankMATCHA