qmk/qmk_firmware #3594: Add support for 00, 000 keycodes

ju0, 2018-08-08:
Some keyboards (commonly 1800 layouts) support the 00 key that types two 0's at the same time. Very rarely a keyboard will support the 000 key, which types three 0's at once. TMK used to support these, but I think QMK has depricated this. I'd like to add two keycodes:

`KC_P00` for a '00'
 #`KC_P000` for a '000' macro

drashna, 2018-08-08:
These are actually supported scan code, no reason not to include them...
http://www.usb.org/developers/hidpage/Hut1_12v2.pdf
Page 57, 2/3 of the way down.

fauxpark, 2018-08-09:
Not ideal, but this might work. `register_code`/`unregister_code` check whether the keycode is one of the media keys (because they need to be sent in a slightly different way), so we just bypass them entirely and do the `add_key`/`del_key`/`send_keyboard_report` dance ourselves.

```c
enum my_keycodes {
  MY_00 = SAFE_RANGE,
  MY_000
};

bool process_record_user(uint16_t keycode, keyrecord_t* record) {
  switch (keycode) {
    case MY_00:
      if (record->event.pressed) {
        add_key(0xB0);
      } else {
        del_key(0xB0);
      }

      send_keyboard_report();

      return false;
    case MY_000:
      if (record_event.pressed) {
        add_key(0xB1);
      } else {
        del_key(0xB1);
      }

      send_keyboard_report();

      return false;
    default:
      return true;
  }
}
```

fauxpark, 2018-08-09:
Update: the above code does work, however it doesn't appear that those keycodes actually do anything on either macOS or Windows. I could get it to print out the numpad `/` with `0x54` but the 00 and 000 keycodes didn't print anything.

I think you may have to use the `SEND_STRING` method instead.

ju0, 2018-08-09:
Yeah, the `SEND_STRING` method is the way to go I think. Then you can have your media keys and sneak in the 00/000 support.

drashna, 2018-08-10:
Ah, yeah, may need to change the keyboard layout in the OS to properly support this, then. 

So .... yeah, macros it is. 

And in that case, I use `tap` actually.  It's a custom function in my user space that calls `register_code` and `unregister_code`.  And just double tap `KP_P0` so it uses the keypad code, and not the num row.

drashna, 2018-10-22:
I'm closing this, as the 00 and 000 keycodes are OS dependant, and were likely removed due to a lack of OS support for them.

closed 2019-03-29 by drashna