qmk/qmk_firmware #3613: Microcontroller recommendation?

jablko, 2018-08-10:
Which microcontroller do you recommend for a new project? I found this list of [compatible microcontrollers](https://github.com/qmk/qmk_firmware/blob/master/docs/compatible_microcontrollers.md). I think the Teensy was once a popular choice. Looking at the repository, out of approximately 251 keyboards, 201 use the atmega32u4. What's your current recommendation?

In particular, what's the best choice for a Bluetooth keyboard? or is the Bluefruit LE [1][2] the only option?

[1] https://www.adafruit.com/product/2633
[2] https://www.adafruit.com/product/2829

drashna, 2018-08-10:
I personally recommend the AT90USB1286 (what the Teensy ++ 2.0 uses).  

It supports a lot more pins, and has a much larger firmware size (120kb vs 28kb).  

Otherwise, if you want to go ARM, then the ST32F303xC chips (the Teensy 3.2 IIRC, and the Planck Rev6)

jablko, 2018-08-19:
Thank you! What do you think about the nRF52? It combines a Cortex-M4 and a BLE radio. There's some [preliminary support](https://github.com/qmk/ChibiOS-Contrib/tree/master/os/hal/ports/NRF5) in ChibiOS.

drashna, 2018-08-22:
Well if ChibiOS supports it, then yeah, definitely.  Though that depends if we're using that version of chibiOS

sypl, 2018-09-08:
What are the benefits of the ARM chips, if I may ask? I'm not that familiar with them.

drashna, 2018-09-08:
@sypl Higher clock frequency (very useful for rgb animations, as that kind of pushes the limits of AVR chips already), larger firmware size (I'm about 100bytes from too much for most of my boards, since they're AVR), and a lot more pins that can be used (larger keyboards, more secondary features, etc)

ishtob, 2018-09-08:
I believe there is support for the nRF52 family, but I think people are still trying to figure out how to get the wireless stacks working smoothly.

You can definitely choose one that has USB and use the nrf as the controller though

jablko, 2018-09-08:
Thank you for this info! The nRF52832 (nor the nRF51822) doesn't have USB. Is there any support in QMK for an add-on USB peripheral, e.g. a USB to UART bridge?

ishtob, 2018-09-09:
yeah, I have been thinking of doing something similar, but decided that a pre-built nrf module as slave paired with a MCU of your choice may actually solve a lot of headache with FCC testing requirements, as long as the module is certified

ishtob, 2018-09-09:
what about nrf52840? I believe it has native USB, but not sure if someone has adopted it in Chibios yet

bricewge, 2018-10-23:
nrf52840 would definitely be a good chip to use with QMK. Someone got ChibiOS to work on it there: https://github.com/netik/dc27_badge/tree/master/software/firmware/ChibiOS-Contrib/os/hal/ports/NRF5/NRF52840

yiancar, 2018-10-23:
you would probably need to get the nordic usb stack working there, dont know how much work would that be

CaninoDev, 2019-08-31:
> Otherwise, if you want to go ARM, then the ST32F303xC chips (the Teensy 3.2 IIRC, and the Planck Rev6).

Why is the STM32F3 series recommended over the STM32F0 series? I thought the F0 series would be better suited because it doesn't require separate crystal components (thus saving space on the PCB and one less point of failure).

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc