qmk/qmk_firmware #3635: UT47 tap dance not working as expected

UltimoKazuma, 2018-08-13:
My issue is somewhat similar to [this one](https://github.com/qmk/qmk_firmware/issues/2162) and [this one](https://github.com/qmk/qmk_firmware/issues/2003). I'm using a UT47 and want to press a key one to send Tab and twice to send Esc. I tested this mainly in keyboardtester.com but also verified it in a text file. What happens is if I press once, it doesn't send anything. But if I press it twice, sends Esc like it should. If I press it and then press another key, it will send Tab and that other key.

The UT47 uses a pro micro, so I don't think there are any hardware limitations going on. The code also compiles fine in msys2.

I just cloned the repo 8 days ago, so I think it should be fairly up to date.

The start of my keymap.c is as follows-

```
#include QMK_KEYBOARD_H
//Tap Dance Declarations
enum {
  TD_TAB_ESC = 0
};
//Tap Dance Definitions
qk_tap_dance_action_t tap_dance_actions[] = {
  [TD_TAB_ESC] = ACTION_TAP_DANCE_DOUBLE(KC_TAB, KC_ESC)
};
```
I am using `TD(TD_TAB_ESC)` in place of a keycode.

I have `#define TAPPING_TERM 200 `in my config.h and `TAP_DANCE_ENABLE = yes` in my rules.mk.

Any help would be appreciated and let me know if I need to clarify anything.

wanleg, 2018-08-18:
What happens if you set up a 'Quad Function Tap-Dance'? just copy the example in the docs exactly and see if it works properly or not?

UltimoKazuma, 2018-08-18:
I get an error when trying to make the file. 

```Compiling: keyboards/ut47/keymaps/UT47_Colemak/keymap.c                                            keyboards/ut47/keymaps/UT47_Colemak/keymap.c:41:3: error: 'CT_CLN' undeclared here (not in a function)
  [CT_CLN] = ACTION_TAP_DANCE_FN_ADVANCED (NULL, dance_cln_finished, dance_cln_reset)
   ^
keyboards/ut47/keymaps/UT47_Colemak/keymap.c:41:2: error: array index in initializer not of integer type
  [CT_CLN] = ACTION_TAP_DANCE_FN_ADVANCED (NULL, dance_cln_finished, dance_cln_reset)
  ^
keyboards/ut47/keymaps/UT47_Colemak/keymap.c:41:2: error: (near initialization for 'tap_dance_actions')
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:360: .build/obj_ut47_UT47_Colemak/keyboards/ut47/keymaps/UT47_Colemak/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:544: ut47:UT47_Colemak:avrdude] Error 1
```

Perhaps I'm misunderstanding how to implement this. This is what I have for the first part of my keymap. I have two enum sections- is that an issue? I'm also not sure if I need to put code before or after the keycodes?

```#include QMK_KEYBOARD_H
enum {
 CT_SE = 0
};
//Tap Dance Declarations
void dance_cln_finished (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
    register_code (KC_RSFT);
    register_code (KC_SCLN);
  } else {
    register_code (KC_SCLN);
  }
}

void dance_cln_reset (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
    unregister_code (KC_RSFT);
    unregister_code (KC_SCLN);
  } else {
    unregister_code (KC_SCLN);
  }
}

//All tap dance functions would go here. Only showing this one.
qk_tap_dance_action_t tap_dance_actions[] = {
 [CT_CLN] = ACTION_TAP_DANCE_FN_ADVANCED (NULL, dance_cln_finished, dance_cln_reset)
};
#ifdef LED_ENABLE
  #include "protocol/serial.h"
#endif

#define _______ KC_TRNS
#define LT3_TAB LT(3, KC_TAB)
#define MT_RSFT_ENT MT(MOD_RSFT, KC_ENT)

enum custom_keycodes {
    LED_TOG = SAFE_RANGE,
    LED_CHG
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

 

LAYOUT(
  TD(CT_SE), KC_Q, KC_W, KC_F, KC_P, KC_B, KC_J, KC_L, KC_U, KC_Y, KC_SCLN, KC_BSPC,
  MO(2), KC_A, KC_R, KC_S, KC_T, KC_G, KC_K, KC_N, KC_E, KC_I, KC_O, KC_QUOT,
  KC_LSFT, KC_X, KC_C, KC_D, KC_V, KC_Z, KC_M, KC_H, KC_COMM, KC_DOT, KC_SLSH, MT(MOD_RSFT, KC_ENT),
  KC_LCTL, KC_LGUI, KC_LALT, TT(1), KC_LSFT, MO(3), KC_SPC, KC_LEFT, KC_DOWN, KC_UP, KC_RGHT
),```

wanleg, 2018-08-19:
can you fork qmk in to your github account so i can see the rest of your `keymap.c` (rather than posting it here)?
or else take a look at [this](https://github.com/wanleg/qmk_firmware/blob/wanlegPersonal/keyboards/xd75/keymaps/wanleg/keymap.c).
You should be able straight-copy from 
`/////////////// TAP DANCE SECTION START ///////////////`
to
`///////////// TAP DANCE SECTION END ///////////////`
and then drop 
`TD(ENT_TAP_DANCE)` into your `keymap.c` instead of `TD(TD_TAB_ESC)`
It should be single-tap = `space`; hold = `shift`; double-tap = `enter`
If that works, then you should be able to start replacing `KC_SPC` with `KC_TAB`, `KC_LSFT` with `KC_NO` and `KC_ENT` with `KC_ESC`

UltimoKazuma, 2018-08-20:
Ok I forked the master and added my folder with my keymap. It has the modified keymap where I copied what you have, and still get the same strange behavior. Simply single/double tapping or holding the key assigned to tap dance doesn't do anything. It only outputs space when I tap it and then tap another key. My fork also includes my keymap where I tried to do the complex TD example.

wanleg, 2018-08-20:
made a pull request to your repo with some testing stuff. let me know how it goes.

UltimoKazuma, 2018-08-21:
Awesome, I got it working! I really appreciate the help. Was it something simple?

wanleg, 2018-08-21:
just like with the issue you mentioned in your [OP](https://github.com/qmk/qmk_firmware/issues/2162), it's gotta be something with backlighting. I don't have a UT47 and don't have any keyboards with backlighting installed/enabled, so this is probably as far as i can help on this.   
If you don't have backlighting, i can probably tweak your keymap so it doesn't load the stuff that makes that part work so that at least TapDance will work for you.   
If you DO have backlighting, someone else would need to assist so you can get both working.  Maybe @filoxo has some insight?

filoxo, 2018-08-21:
You shouldn't need to remove anything if you're not using LED backlighting, that is already guarded for (see the [UT47 README](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ut47#instructions)). 
The only thing I can think that *might* be problematic is the way its communicating LED state to the second controller (uses serial). See this line for how its sending that: https://github.com/qmk/qmk_firmware/blob/master/keyboards/ut47/ut47.c#L40
It would only be doing that if you have LEDs but shouldn't be preventing the default actions for features like TD. 
I can dig into the PR that wanleg sent to see if I missed something.

wanleg, 2018-08-21:
i only made a board WITHOUT any kind of backlight support. the issue on a different board (jj40) was to do with how matrix scanning was done, i think. 
If you have a UT47 still, do you have any issues with TapDance and `LED_ENABLE=yes`? 
I wonder if there's something going on where the communication with the second promicro interacts in some weird way with TapDance interrupts.
Or maybe it's somehow related to something [like this](https://www.reddit.com/r/olkb/comments/8en8f1/strange_caps_lock_behavior/)? like something with the led.c setup...

filoxo, 2018-08-21:
Sure I can see if I can play around with mine. I've also seen a report of issues with RGB underglow if also used with LED backlight (one of the LED indicator pins needs to be abused for something like this, and I can't seem to find the thread I was thinking of). So I suspect it is something to do with the serial communication to the other pro micro.

UltimoKazuma, 2018-08-21:
I personally don't have backlighting, so if you feel like editing the keymap, that's cool. I don't really need it though, since I'm fine with how it behaves currently, so don't worry about it unless it's a really quick thing.

wanleg, 2018-08-22:
@UltimoKazuma, in your `ut47/keymaps/UT47_Colemak/rules.mk`, i _think_ you can replace 
```
# custom matrix setup
CUSTOM_MATRIX = yes
SRC += matrix.c protocol/serial_uart.c  
```   
with 
```
# custom matrix setup
ifdef CUSTOM_MATRIX
  CUSTOM_MATRIX = no
endif
```
to get the same effect. or else maybe drop that in the `ut47/rules.mk` if that doesn't work.  
not 100% sure that'll disable the problem part though.

drashna, 2018-08-22:
@wanleg the ifdef part won't work.   And it causes issues if you try to disable the custom matrix. I've tried.

drashna, 2018-10-22:
Was this still an issue?

wanleg, 2018-10-23:
sounds like @UltimoKazuma will just use the alternate setup i put in his repo and @filoxo was going to do some digging but hasn't replied back here. I don't have a UT47 to do more in-depth work.
if there's a workaround, close issue?

closed 2018-10-23 by drashna