qmk/qmk_firmware #3653: Inavlid hex file after compilation for teensy (atmega32u4)

TheEdgeOfRage, 2018-08-14:
I handwired a keyboard around a year ago and had no issues with QMK until today. What I usually did was run

`make handwired/<keyboard>:<keymap>`

and then

`teensy-loader-cli -w -v -mmcu=atmega32u4 output.hex`

to flash the firmware. But today after successfully compiling the firmware, teensy-loader-cli returns the following error when trying to flash it:

    Warning, HEX parse error line 1372
    error reading intel hex file "handwired_ergomini_raging.hex"

The teensy loader software hasn't been updated since 2016, so I guess something changed in QMK. But then again, I tried to check out an older commit that I know for sure worked only to be met with the same error again.

When I open the hex file, the end looks like this, which doesn't look right:

    1364 :105530003A183118311841183A183A1841185325B9
    1365 :105540007B2564267B256426AE25CF25642626266A
    1366 :02555000322601                                                                                                                                                      
    1367 :1055520011010102DB01B01F2420B31FEF1FB41F92                                                                                                                          
    1368 :10556200017465737470617373003031323633C6FF                                                                                                                          
    1369 :1055720037909192936695F1F4F5F6F74036392813
    1370 :0C55820007F206B90665067E06970600D3
    1371 :0200000400807A
    1372 :04027F00000000007B
    1373 :00000001FF

So in the end:

https://i.imgur.com/OCUy2zP.png

As I have no idea where even to start solving this. The code for my keyboard can be found here:

https://github.com/TheEdgeOfRage/qmk_firmware/tree/ergomini/keyboards/handwired/ergomini

and if there's anything else I can help with just let me know.

Thanks

drashna, 2018-08-14:
At this line: 
https://github.com/TheEdgeOfRage/qmk_firmware/blob/ergomini/keyboards/handwired/ergomini/rules.mk#L47

Try replacing it completely with `BOOTLOADER = halfkay`, and see if that helps. 

I doubt it, but ...

TheEdgeOfRage, 2018-08-14:
That actually worked. Thanks :)

closed 2018-08-14 by TheEdgeOfRage

drashna, 2018-08-14:
Awesome!!  Glad to hear it!