qmk/qmk_firmware #3657: avrdude flashing error on AVR GCC 8.2.0

raetiacorvus, 2018-08-15:
Hi. At least on arch linux with avr-gcc 8.2.0 compiling a fresh firmware and trying to flash it with avrdude results in the following error:

```
avrdude: ERROR: address 0x800293 out of range at line 1027 of ./.build/olympia_carrera_2ter.hex
```
Downgrading to avr-gcc 8.1.0 resolves the issue.
It seems other people are experiencing the same problem: https://www.reddit.com/r/olkb/comments/96r0os/lets_split_vitamins_included_issues/

alesguzik, 2018-08-20:
Same problem on NixOS. Downgrading gcc to 8.1.0 fixed it for me.

raetiacorvus, 2018-08-21:
so far i was able to bisect it to the following PR/commit https://github.com/gcc-mirror/gcc/commit/db7548a2771bbf34cf7430712af7ac670b429958#diff-10d8692c1cc5c8611aa7599f107f0f56

raetiacorvus, 2018-08-22:
I probably have no idea what i am doing but i hope the info i can gather helps.
Getting the symbols from the `elf` with `avr-nm -n keyboard.elf` shows the following changes when building the firmware before the problematic commit and after.
```
589,591c589,591
< 0080029c B reset_key
< 008002a0 B _end
< 00810000 N __eeprom_end
---
> 0080029c D reset_key
> 008002a0 D _end
> 00810000 D __eeprom_end
```
from `avr-nm` man page:

> "B" "b" The symbol is in the BSS data section.  This section typically contains zero-initialized or uninitialized data, although the exact behavior is system dependent.
> "D" "d" The symbol is in the initialized data section.
> "N" The symbol is a debugging symbol.

So it looks like `reset_key` and `_end` both changed from the BSS data section to the data section which results in the `0x800293 out of range` error.
As far as i can tell `reset_key` is set in the following line to the `.noinit` section which should be part of the BSS data section.
https://github.com/qmk/qmk_firmware/blob/6076ed85be111f0e88c7049aa3d2e69eed5005a1/tmk_core/common/avr/bootloader.c#L73

As far as i can tell qmk doesn't use a custom linker script but the default one installed with the `avr-binutils` package in `/usr/lib/ldscripts/`. There is a custom one in the tmk folder ( `qmk_firmware/tmk_core/ldscript_keymap_avr5.x) but it isn't used.

raetiacorvus, 2018-08-23:
It looks like the `.noinit` section gets changed from `NOBITS` to `PROGBITS`
```sh
Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .data             PROGBITS        00800100 004706 00005a 00  WA  0   0  1
  [ 2] .text             PROGBITS        00000000 000094 004672 00  AX  0   0  2
  [ 3] .bss              NOBITS          0080015a 004760 000142 00  WA  0   0  1
  [ 4] .noinit           NOBITS          0080029c 004760 000004 00  WA  0   0  1
…
```
```sh
Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .data             PROGBITS        00800100 004718 00005a 00  WA  0   0  1
  [ 2] .text             PROGBITS        00000000 0000b4 004664 00  AX  0   0  2
  [ 3] .bss              NOBITS          0080015a 004772 000142 00  WA  0   0  1
  [ 4] .noinit           PROGBITS        0080029c 004772 000004 00  WA  0   0  1
…
```

raetiacorvus, 2018-09-06:
The issue has been fixed in binutils master https://sourceware.org/bugzilla/show_bug.cgi?id=23570

drashna, 2018-09-12:
Awesome!  
So for now, you should revert to 8.1, until a new version is pushed to package managers, correct?

hendersonreed, 2018-10-12:
I just wanted to comment that the hackish workaround posted by @raetiacorvus has been totally successful for me. So, if like in my case your distribution doesn't support installing a specific outdated version, you can use that until the new version is pushed to the repositories.

closed 2018-10-22 by drashna

mvaneijgen, 2019-03-04:
I'm having almost the same error, but not fully. 

`avrdude: ERROR: address 0x8001ef out of range at line 1398 of ~/Documents/qmk_firmware/.build/mvaneijgen_numpad_split_default.hex`

While searching for a solution I came across this thread, but I am on gcc version 8.3

```
brew info gcc
gcc: stable 8.3.0 (bottled), HEAD
GNU compiler collection
https://gcc.gnu.org/brew info gcc
```

Does this issue still persist for version 8.3? Or should I make a new thread?

kaylanm, 2019-03-04:
`gcc` is not the same as `avr-gcc`, but avr-gcc@8.3 is still affected.