qmk/qmk_firmware #3661: STM32 EEPROM emulation library and licensing

yiancar, 2018-08-15:
Hello everyone:)

As some of you know using STM32 is the new trend but qmk lacks some features for that family of processors. Today I want to focus on the eeprom emulation as everyone likes some memory on there keyboards.

The industry standard method is using the ST standard library (https://www.st.com/en/embedded-software/stsw-stm32061.html). This however comes with the "Mix Ultimate Liberty+OSS+3rd-party" license (https://my.st.com/content/ccc/resource/legal/legal_agreement/license_agreement/group0/39/50/32/6c/e0/a8/45/2d/DM00218346/files/DM00218346.pdf/jcr:content/translations/en.DM00218346.pdf). Mostly have a look at paragraph 5 and 6 which limit the use of the library under open source terms.

The main solution I have seen is adding an exception in our license to say that the specific file is not open source and that we will use it as is, which is allowed.

There is also other libraries (https://github.com/nimaltd/EEPROM) which however I question as the manufacturers one should be superior.

I am not a lawyer or a software licensing expert so please give your input. However a solution for this must be found!

drashna, 2018-08-15:
@jackhumbert

yiancar, 2018-08-15:
An example can be this : https://github.com/d-ronin/dRonin/issues/928

yiancar, 2018-08-17:
@fredizzimo I hear u got some ideas?:)

yiancar, 2018-08-23:
Ok I have tried multiple things.
- The library (ST one) suggested by @fredizzimo . You can find this here : https://github.com/yiancar/qmk_firmware/tree/eeprom .
- The maple libraries ( targeted for arduino ide but i remodified the code) which can be found here : https://github.com/rogerclarkmelbourne/Arduino_STM32 and my implementation here: https://github.com/yiancar/qmk_firmware/tree/eeprom_maple .

None of them seem to work and they hang the keyboard. I dont know If its because it is trying to read and write at the same time on flash, I never managed to hookup a debugger on one of the arm powered keyboard, so I can only do basic checks.

I am very tempted to just hook up an i2c eeprom and call it a day at this point. If anyone is interested or has any experience with this, please point me in the right direction

yiancar, 2018-08-24:
UPDATE:
A working solution has been implemented here. This assumes that the final 4kb of the flash memory are unused by the program and are use by emulation. The chibios linker script can be modified to ensure this however it is located within chibios currently. Furthermore the library is currently implemented for an STM32F303CC which has 256kb of flash. For bigger/smaller STM32F3XX the header file must be changed to the correct flash size.

ishtob, 2018-08-31:
tested working on the STM32F303CB on the planck rev6, anyone has other stm32 chips they can test it on and confirm those are working too?

yiancar, 2018-09-05:
Any stm32 with 256kb of flash should work. For the rest people should adjust the .h file which is pretty easy

closed 2018-09-10 by drashna