qmk/qmk_firmware #3663: MacOS alt (fn+option) keycode

calvn, 2018-08-15:
I've been trying to find the keycode for the `alt` key press (`fn+option`), but could not find it in the [docs,](https://docs.qmk.fm/#/faq_keymap), GH issues, or internet forums. `KC_LALT` acts as `option` on MacOS, but I would like to have a key mapped for `alt`. There's an open discussion in #2179 with regards to the `fn` key, but it seems that mapping the `fn` key is not recommended so there's no straightforward way to register that key press.

Wondering if there's a keycode for it that's documented elsewhere, and if not what would be the best approach to map that to my keymap?

egla89, 2018-08-15:
fn + option does nothing on a Mac. Alt is just written on top of it to ensure compatibility with non MacOS software.

> Since the 1990s, "alt" typically appears on the key, as well, for use as an Alt key with non-Mac software, such as Unix and Windows programs

calvn, 2018-08-15:
They appear to register different escape sequences through key combinations:

fn+option + up: ^[[5~
option + up: ^[[1;5A

I currently can't do the former in my external keyboard.

egla89, 2018-08-15:
were do you see those? Even MacOS itself shows me no difference

egla89, 2018-08-15:
option + up is just up on my mac
fn + up = page up
fn + option + up = page up

Go to system preferences -> keyboard -> Input Sources -> show input in menu bar and then "Show keyboard viewer"

calvn, 2018-08-15:
You can see the escape sequence of your input in the terminal if you do `ctrl+v` and then perform the keypress combo.

egla89, 2018-08-15:
can you confirm that

> fn + option + up

has a different result than

> fn + up

?

calvn, 2018-08-15:
Yes, they are different. Over the terminal `fn + up` simply scrolls up the terminal. `fn + option + up` is mapped to a command on my zsh config, which is the escape sequence for `^[[5~`.

In the keyboard viewer, you can also see the difference if you keep `option` held and then press `fn` a couple of times to see that a few keys are different -- namely the arrow keys, enter, and backspace.

egla89, 2018-08-15:
well if you mapped command to a shortcut that obviously explains it.

calvn, 2018-08-15:
This is regardless of what I mapped on zsh (it's an escape sequence to command, not a keypress to escape sequence mapping). I could have an empty zsh config, or not be using zsh at all for this matter, and the above still stands. The actual keypress registers differently.

skullydazed, 2018-08-15:
@calvn I think if you press Alt+Page Up on your external keyboard you will get the same result as Fn+Alt+Up on your laptop. Can you give that a try?

calvn, 2018-08-15:
@skullydazed I don't have the page up keycode mapped. I'll flash a new keymap with that on it later today and give it a try. Thanks for the suggestion :)

calvn, 2018-08-15:
From the looks of the icons in the keyboard viewer it seems that the addition of `fn` turns those different keys into map to KC_DEL, KC_HOME, KC_END, KC_PGUP, KC_PGDN, and KC_PENT, so that's promising.

calvn, 2018-08-16:
@skullydazed That was it! Doing a `Alt+Page Up` and `Alt+Page Down` sends `^[[5~` and `^[[6~` respectively.

closed 2018-08-16 by calvn