qmk/qmk_firmware #3680: JJ40 + Userspace + mousekeys = nothing

wanleg, 2018-08-17:
Came across an issue on my jj40 with my userspace and mousekeys - it doesn't work.
my userspace and mousekeys works fine with a Let's Split, Gherkin, and xd75 (don't have anything else to test with).
Using a qmk_firmware/keyboards/jj40/keymaps setup with mousekeys also works fine.

qmk_firmware/layouts/community/ortho_4x12/crs also has mousekeys in it, so i flashed that on the jj40 and it doesn't work either. no problem on other boards, just like with my layout.

any ideas on what keyboard-local configs might be doing that Userspace configs aren't?

again, this problem seems specific to the jj40

drashna, 2018-08-22:
I think you need to remove the ifndef's in the userspace's rules.mk.  

crs' keymap has the keys, but the rules.mk doesn't set it. So it's defaulting to the keyboard

wanleg, 2018-08-22:
!
thanks for taking the time to correct my noob mistake! was so busy mucking around in /users and /layouts...

closed 2018-08-22 by wanleg